Joshua-Riek/ubuntu-rockchip #708: Orange Pi 5 not able to config 24.04 beta

FreeWilly2022, 2024-04-13:
I did some testing of the Ubuntu 2404 beta on the OPi 5 8 gb with SSD. I downloaded the image to an SD card and booted the OPi 5 with it. It started up normally, configured the keyboard and found my USB Wi-Fi. After I entered the info for the Wi-Fi the config window went away. The mouse pointer continued moving but nothing was on the screen. I rebooted and it gave me a quick message that it was going into a back up mode, and then let me configure the log in. At this point Chrome did not work. I tried doing the command to flash U-Boot to the SPI, but the file was not found (I checked the /usr/lib/ directory and the file was not there).

I did another reboot and now Chrome is working well at 4k. 

I can upload the syslog if that would help you. Thanks for putting this together!

Joshua-Riek, 2024-04-13:
It sounds like you have a corrupt install, are you able to download the image again and re-flash your SD card? 

> I tried doing the command to flash U-Boot to the SPI, but the file was not found (I checked the /usr/lib/ directory and the file was not there).

Please read the new documentation on how to install U-Boot to the SPI:
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04#install-u-boot-to-the-spi-flash

Joshua-Riek, 2024-04-13:
Because of the installer issues I'm removing the Desktop image for the Ubuntu 24.04 Beta. I have to wait a few days for Ubuntu to update their packages, I can not do it manually myself.

closed 2024-04-13 by Joshua-Riek