Joshua-Riek/ubuntu-rockchip #710: Ubuntu 24.04 Desktop Installer Crashing

Joshua-Riek, 2024-04-13:
It seems the desktop installer will crash during setup. 

Logs from `/var/log/oem-config` show an error with libsoup.
```
(oem-config:1073): libsoup-ERROR **: 01:39:06.095: libsoup3 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.
```

JFLim1, 2024-04-13:
The desktop installer also crash during setup when I first boot up. During first boot, the initial set up did not complete, it basically stuck or crash.  

Had to power cycle to restart and similar situation after reboot and set up was incomplete but this time the set up system automatically reboot and allow me to complete the system set up.

After the initial setup completed successfully, Chromium will NOT launch -- this is before reboot.  But Chromium works as expected after "reboot".

Joshua-Riek, 2024-04-13:
I'm trying to fix this now, I think there is an issue with ubiquity, it was recently updated to use libsoup3.0, but some other libs still depend on libsoup2 causing conflict. 

https://launchpad.net/ubuntu/+source/ubiquity/24.04.4
http://launchpadlibrarian.net/723838578/ubiquity_24.04.3build3_24.04.4.diff.gz

Joshua-Riek, 2024-04-13:
The mirror http://ports.ubuntu.com, has not been synced yet and is missing some package updates. The below package ubiquity-frontend-gtk (which is part of ubiquity) should be at version 24.04.4 to solve this issue. 
 
```
ubuntu@localhost:~$ apt search ubiquity-frontend-gtk
Sorting... Done
Full Text Search... Done
ubiquity-frontend-gtk/noble,now 24.04.3 arm64 [residual-config]
  GTK+ frontend for Ubiquity live installer

ubiquity-frontend-gtk-panel/noble 24.04.3 arm64
  GTK+ frontend panel implementation for ubiquity-dm UI
```

Joshua-Riek, 2024-04-13:
I would copy the package to my PPA to fix this, but since they are pinned it may cause an issue. I need to do some testing.

Joshua-Riek, 2024-04-13:
I may remove the desktop images from the beta release, we need to wait for ubuntu to update their package servers, there are to many for me to port over myself to fix the installer issue.

closed 2024-04-15 by Joshua-Riek