Joshua-Riek/ubuntu-rockchip #711: LibreOffice Will Not Launch

JFLim1, 2024-04-13:
LibreOffice will not launch either by click on the "LibreOffice Writer" icon or launch from terminal.

```
jfl@localhost:~$ libreoffice --writer
Warning: failed to launch javaldx - java may not function correctly
ERROR 4 forking process
jfl@localhost:~$ libreoffice
Warning: failed to launch javaldx - java may not function correctly
ERROR 4 forking process
jfl@localhost:~$ 
```

Joshua-Riek, 2024-04-13:
May be a side effect of the desktop installer not working as the system can not properly setup.

Joshua-Riek, 2024-04-13:
Because of the installer issues I'm removing the Desktop image for the Ubuntu 24.04 Beta. I have to wait a few days for Ubuntu to update their packages, I can not do it manually myself.

closed 2024-04-13 by Joshua-Riek

JFLim1, 2024-04-14:
Hi @Joshua-Riek,

FYI, found a temporary solution to launch LibreOffice from command line:

`/usr/lib/libreoffice/program/soffice.bin`

JFLim1, 2024-04-16:
Hi @Joshua-Riek,

A feedback.  Successfully install the new Ubuntu 24.04 v2 beta1 30 minutes ago.  The installer crash seems to be resolved with this new image.

LibreOffice will still refuse to launch by click on the "libreoffice icon" on the panel.  Same error:

```
jfl@jfl-opi5plus:~$ libreoffice
Warning: failed to launch javaldx - java may not function correctly
ERROR 4 forking process
```
Will launch with:

`/usr/lib/libreoffice/program/soffice.bin
`