Joshua-Riek/ubuntu-rockchip #715: Orange Pi 5 Plus Beta crash loops during boot

ovalhub, 2024-04-14:
- I downloaded the [24.04-beta for Orange Pi 5 Plus](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.0.0-beta/ubuntu-24.04-beta-preinstalled-server-arm64-orangepi-5-plus.img.xz) image, use balenaEtcher to burn it to SD card, booted computer, all good, booted from SD card board is stable
 - then I copied the same beta archive to the SD via scp so that I can copy it to the eMMC at /dev/mmcblk0: 
   . `sudo dd bs=1048576 if=/dev/zero of=/dev/mmcblk0 count=2048 status=progress`
   . `lz4 -dc /mnt/backups/dev.mmcblk0 | pv -s 233G | sudo dd conv=sparse of=/dev/mmcblk0 bs=4k`
   . `sync; sudo halt`
 - then I remove the SD card and reboot

The computer reboots but not just once, sometimes up to 3 times. It eventually boots up and quiets down but the SSD server refuses connections and crashes.

I had backed up my eMMC before:
  - `dd if=/dev/mmcblk0 bs=4M | pv -s 233G | lz4 -c --fast > /mnt/backups/dev.mmcblk0`
and restored it with:
  - `lz4 -dc /mnt/backups/dev.mmcblk0 | pv -s 233G | sudo dd of=/dev/mmcblk0 bs=4k`
and it restored fine, board is booting and operating normally as before running 22.04

Joshua-Riek, 2024-04-14:
Uh can you provide logs? I cannot do anything with the provided information.

I've been using the Orange Pi 5 Plus as my primary testing device and have not encountered anything you say. However, there is a chance the SPL is not matching causing ram instability, if you provide uart boot logs I can see if this is the problem.

ovalhub, 2024-04-14:
> Uh can you provide logs? 
The system is unstable so I don't know how to provide any logs, I have no way to extract anything, sshd is unoperative.
> if you provide uart boot logs I can see if this is the problem
I don't know how to do this, if you have instructions you can point me at I may be able provide logs. I believe I even have a UART debugging device I never used lying around someplace. 
I use an old VGA LCD display with HDMI->VGA adapters for OrangePI or RaspberryPI to connect display and a USB keyboard for network access debugging, but without ssh access, I feel a little helpless. Happy to learn with instructions (!)

Joshua-Riek, 2024-04-15:
I need some time to document how to troubleshoot this sort of problem as it has been encountered before. I'll write up a document tomorrow night.

ovalhub, 2024-04-15:
> > The computer reboots but not just once, sometimes up to 3 times. It eventually boots up and quiets down but the SSH server refuses connections and crashes.
> 
> Have you checked that your power supply is reliable?

I've been running 22.04 and 24.04-beta with my iPhone charger (the larger kind) for the last couple of weeks because my original power supply died. I now received a 27W, Raspberry PI 5 compatible, power supply to replace the original and I'm running 22.04 reliably with it.
To be sure it wasn't the iPhone charger being too weak for 24.04-beta or something (that would be an interesting datapoint too since it runs 22.04 reliably), let me try 24.04-beta with the new power supply and report back...

ovalhub, 2024-04-15:
Good call on the power supply !
I was able to setup 24.04-beta using the new 27W power supply, system is stable.

Now onto why I can't start my VMs:
```
$ virsh create /mnt/backups/2024-04-13/oluf.xml
error: Failed to create domain from /mnt/backups/2024-04-13/oluf.xml
error: internal error: QEMU unexpectedly closed the monitor (vm='oluf'): 2024-04-15T19:56:48.374456Z qemu-system-aarch64: Failed to put registers after init: Invalid argument
```
But this problem unrelated to this issue. Closing as resolved. Thank you again for the hint !

closed 2024-04-15 by ovalhub

Joshua-Riek, 2024-04-15:
> Good call on the power supply ! I was able to setup 24.04-beta using the new 27W power supply, system is stable.
>
> Now onto why I can't start my VMs:
>
> ```
> $ virsh create [...]

Cool! As for the other issue i found the below bug report that may be able to help:

https://issues.redhat.com/browse/RHEL-21761

ovalhub, 2024-04-16:
> Cool! As for the other issue i found the below bug report that may be able to help:
>
> https://issues.redhat.com/browse/RHEL-21761

Hmm, not sure. I'm able to run my `qemu-system-x86` VM without a problem but all my `qemu-system-aarch64` ones fail the same way.
The [21761](https://issues.redhat.com/browse/RHEL-21761) is marked dup of another I don't have access to. I guess, I need to wait for more people to hit this issue... Anyhow, this is off topic here, sorry for the noise.