Joshua-Riek/ubuntu-rockchip #716: Upgrade 22.04 => 24.04

smjltd, 2024-04-15:
Can we expect a comprehensive guide for upgrading Ubuntu from 22.04 to 24.04, or is it as straightforward as using the upgrade command?

Joshua-Riek, 2024-04-15:
No, there are too many system level hacks that will completely break the system. 

I've been working hard so this will not be the case in the future. However, I'm limited in my time as this is a hobby project, and most bugs that are found will likely require a more complicated solution with extensive testing.

I'm also struggling as I cannot include some features in the desktop releases due to file size limitations for GitHub releases. I do not want to pay to host downloads on some cloud service, but sadly this will likely be what happens.

KlimRider, 2024-04-15:
How about to create some mirror on GDrive? I have enough space on the GDrive

Joshua-Riek, 2024-04-15:
I'm going to need 100GB for each release and this will only grow larger...

KlimRider, 2024-04-15:
For now I can provide 150 Gigs of space. I can increase this to 2Tb

KlimRider, 2024-04-15:
AWS S3 on my opinion very uncommon license. Its hard to predict the volume of license

Joshua-Riek, 2024-04-15:
I see, I'm not familiar with cloud hosting. I still need to do a lot of research. But I wonder if I can split artifacts in half then have a download website bring them together into one file.

KlimRider, 2024-04-15:
You may keep all sources on the GitHub, but all images move to the cloud. On the GitHub publish URLs to the image download. I see this that way. But I don't know how this will be correlated with build feature. May this cause to move images by hand.

bagong, 2024-04-15:
> How about to create some mirror on GDrive? I have enough space on the GDrive. Or we can create another separate account for this

Gdrive download quotas will make you and the downloaders unhappy. The Xulong gdrive images used to be constantly blocked with "quota exceeded" error messages. There are also security issues: if the images are transparently generated by Github actions and downloaded from there, nobody will worry about build authenticity and integrity.

okonon, 2024-04-15:
@Joshua-Riek I do AWS consulting for living and can assure you that AWS S3 is the cheapest cloud object store provider. I created a quick estimate for storage 

I would architect it in this way:
Put a CloudFront CDN distribution in front of S3 bucket and serve your contents from CloudFront that way it is cached closest to your users and S3 is not hit with millions download requests. So essentially you pay for S3 storage and CloudFront will be charged based on user requests. Since I don't know your storage and traffic requirements it is touch to estimate what it would cost. Also we can setup lifecycle policies to move infrequently used objects to cheaper storage tier (say old image is moved to infrequent access tier after 30 days of inactivity, then after 30 more days moved to infrequent access in single Availability zone, and they finally moved to deep archive storage).

I do this for work and i can say that this is the cheapest you can set up such file hosting. I am happy to help you and set it all up if you are interested. Let me know and I can provide my email so we can jump on a call to discuss further. 

Also I created a simple estimate for above setup that you can tweak to see pricing changes:
> CloudFront 1TB Free Tier
> Effective December 1, 2021, the first 1TB per month of data transfer out to the internet and 10 million http/https requests > from all [Edge locations](https://aws.amazon.com/cloudfront/features/) will be free

Pricing examples:
Free Tier:
1 TB of data transfer out of your CDN with 10 Million requests = $0 /month
Additional (on top of free tier):
1 TB with 10 Mil requests (2 TB/month + 20 Mil requests/month) = $97/month
S3 Storage:
Free 5 GB/month free
Additional: 1 TB /month $28.56 (highest storage tier that can be optimized with lifecycle policies)

Here is the estimate that you can play around with:
https://calculator.aws/#/estimate?id=1e7a4c1a7d266c5afceec6134534e257d819d872

SuperKali, 2024-04-15:
I think that wasabi is the most cheaper 😅 not AWS
Wasabi offers S3 storage with 1TB at 6.99$ monthly without any fee on traffic

SuperKali, 2024-04-15:
Every TB is charged at 6.99$

Joshua-Riek, 2024-04-15:
Thanks for the insight, but i do need time to get Ubuntu 24.04 out the door and work on bugfixes. The requirement of having cloud hosted downloading is extra, while it would be nice its not essential.

okonon, 2024-04-15:
Oh thats cool never heard of them. Sorry about that. Let me know if you need any help with AWS.

closed 2024-04-15 by Joshua-Riek