Joshua-Riek/ubuntu-rockchip #718: Orange Pi 5 Plus - unexpected reboots

homonto, 2024-04-15:
I just installed the newest Ubuntu  [ubuntu-24.04-beta-preinstalled-desktop-arm64-orangepi-5-plus.img.xz](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.0.0-beta/ubuntu-24.04-beta-preinstalled-desktop-arm64-orangepi-5-plus.img.xz)

and I am getting all the time reboots
not related to any activity
once on YT
once on settings
once on browsing files

Joshua-Riek, 2024-04-15:
Most crashing or unexpected reboots are caused by two things:
1. Bad power supply, you need static 5v 4a 
2. DDR and SPL blob mismatch causing RAM instability, please update SPI with the command `sudo u-boot-install-mtd` and reboot.

If you can provide system logs i can take a closer look.

homonto, 2024-04-16:
I installed server edition and problem is gone
so it seems the desktop edition has issues
maybe with graphics?
server works like a charm - already installed 20 containers and not finished yet

Joshua-Riek, 2024-04-16:
5v 10a sounds like a problem.. use 5v 4a.

homonto, 2024-04-16:
> 5v 10a sounds like a problem.. use 5v 4a.

brother, are you sure you meant what you wrote?
do you understand what 5V/10A means?
I doubt reading your answer... ;-)

Joshua-Riek, 2024-04-16:
Yes, it sounds like a PD power supply, these boards do not work well with them.

homonto, 2024-04-16:
no, 5V/10A for Power Supply means: "I am delivering 5V up to 10A - no more but anything from 0A to 10A I can deliver easily".

PD is totally different piece of cake - it is universal power supply that delivers certain voltage based on load request, and this voltage can be from 5V up to 19V, provided the load is active and tells the PS what voltage it expects from PS.

So again: my PS is purely 5V but with the max current up to 10A.

Joshua-Riek, 2024-04-16:
Using a 5v 4a power supply is the most reliable for these devices.

If you can provide logs of the crash, I can take a look. Otherwise, adhear to my previous recommendations, and please do not waste my time.

closed 2024-04-16 by homonto