Joshua-Riek/ubuntu-rockchip #719: 24.04 Beta - ubuntu-rockchip-install: command not found

FreeWilly2022, 2024-04-15:
Thanks again for making this release. The good news is that I put this image:
ubuntu-24.04-beta-preinstalled-desktop-arm64-orangepi-5.img.xz
on a SD card and my OPi5 8Gb booted with no problems. 
The installer detected my USB Wi-Fi card:
Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
during installation. 

I Installed U-Boot to the SPI Flash:
sudo u-boot-install-mtd

After rebooting I have been watching YouTube for a few hours, it does 4k with 1/1,000 dropped frames. YouTube commercials play without a delay. The computer woke up after going to sleep with no problems.

The only issue I have is that I could not use the standard command for copying over to my SSD:
$ sudo ubuntu-rockchip-install /dev/nvme0n1
sudo: ubuntu-rockchip-install: command not found

Is there a different command to copy from SD card to SSD?

Thanks

Joshua-Riek, 2024-04-16:
The `ubuntu-rockchpi-install` command has not yet been ported over yet due to changing the filesystem partition layout. I need some time to fix it.

FreeWilly2022, 2024-04-16:
No worries, I just wanted to make sure I wasn't missing something.

closed 2024-04-27 by Joshua-Riek