Joshua-Riek/ubuntu-rockchip #720: MPV Lost VPU HW Acceleration Sometimes While Streaming Youtube - Ubuntu 24.04 Beta1

JFLim1, 2024-04-16:
Hi @Joshua-Riek,

Device: Opi5-Plus

MPV at times will drop vpu hw acceleration streaming Youtube.  It started with vpu acceleration and after a while vpu hw acceleration is no longer available.  

Dropped from "VO: [gpu] 3840x2160 drm_prime[p010]" to "VO: [gpu] 3840x2160 yuv420p10"

```
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=k7dy1B6bOeM
 (+) Video --vid=1 (*) (vp9 3840x2160 59.940fps)
 (+) Audio --aid=1 --alang=eng (*) (eac3 6ch 48000Hz)
File tags:
 Uploader: 4K Video ULTRA HD
 Channel_URL: https://www.youtube.com/channel/UCRauo6y9tx0FB0OLLtqLA-g
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
AO: [pipewire] 48000Hz 5.1(side) 6ch floatp
VO: [gpu] 3840x2160 drm_prime[p010]
AV: 00:01:59 / 00:11:11 (18%) A-V:  0.000 Dropped: 19 Cache: 42s/150MB
Error while decoding frame (hardware decoding)!
Error while decoding frame (hardware decoding)!
Error while decoding frame (hardware decoding)!
Attempting next decoding method after failure of vp9_rkmpp-rkmpp.
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
Error while decoding frame!
[ffmpeg/video] vp9: Not all references are available
[... 152 lines cut ...]
Error while decoding frame!
Error while decoding frame!
Error while decoding frame!
Error while decoding frame!
Error while decoding frame!
AV: 00:01:59 / 00:11:11 (18%) A-V:  0.000 Dropped: 19 Cache: 41s/148MB
Error while decoding frame!
Error while decoding frame!
Error while decoding frame!

Audio/Video desynchronisation detected! Possible reasons include too slow
hardware, temporary CPU spikes, broken drivers, and broken files. Audio
position will not match to the video (see A-V status field).

AV: 00:01:59 / 00:11:11 (18%) A-V:  4.055 Dropped: 19 Cache: 41s/150MB
[ffmpeg] Impossible to convert between the formats supported by the filter 'mpv_src_default_in' and the filter 'auto_scale_0'
[lavfi] failed to configure the filter graph
Disabling filter scale_rkrga.00 because it has failed.
AV: 00:01:59 / 00:11:11 (18%) A-V:  4.138 Dropped: 20 Cache: 41s/150MB
VO: [gpu] 3840x2160 yuv420p10
AV: 00:03:18 / 00:11:11 (30%) A-V:  0.661 Dropped: 2452 Cache: 42s/150MB

Exiting... (Quit)
jfl@jfl-opi5plus:
```

Joshua-Riek, 2024-04-21:
Are you able to reproduce this consistently? I'm testing it out and so far playing the video you linked for 5 minutes with no issues.

JFLim1, 2024-04-22:
> Are you able to reproduce this consistently?

Up until yesterday even after updated/upgraded to the newer mesa-panfork-1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble1, still encountered the random drop of vpu hw acceleration on mpv.  The point or time when the vpu hw acceleration is loss is not fix.  So at times the vpu hw acceleration was loss in less than 2 minutes at times loss after 8 minutes.  Random.

But this morning when I performed system update/upgrade, kernel-6.1.0.1008.8, the mpv seems to behaving correctly streaming various videos without loosing vpu hw acceleration.  In total close to 2 hrs of videos. 

Will continue to test mpv again latter.

Thanks for the great work and support.

```
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=k7dy1B6bOeM
 (+) Video --vid=1 (*) (vp9 3840x2160 59.940fps)
 (+) Audio --aid=1 --alang=eng (*) (eac3 6ch 48000Hz)
File tags:
 Uploader: 4K Video ULTRA HD
 Channel_URL: https://www.youtube.com/channel/UCRauo6y9tx0FB0OLLtqLA-g
Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Could not dynamically load CUDA
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
AO: [pipewire] 48000Hz 5.1(side) 6ch floatp
VO: [gpu] 3840x2160 drm_prime[p010]
AV: 00:11:11 / 00:11:11 (100%) A-V:  0.000 Dropped: 14 Cache: 0.0s

Exiting... (End of file)
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=k7dy1B6bOeM
 (+) Video --vid=1 (*) (vp9 3840x2160 59.940fps)
 (+) Audio --aid=1 --alang=eng (*) (eac3 6ch 48000Hz)
File tags:
 Uploader: 4K Video ULTRA HD
 Channel_URL: https://www.youtube.com/channel/UCRauo6y9tx0FB0OLLtqLA-g
Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Could not dynamically load CUDA
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
AO: [pipewire] 48000Hz 5.1(side) 6ch floatp
VO: [gpu] 3840x2160 drm_prime[p010]
AV: 00:11:11 / 00:11:11 (100%) A-V:  0.000 Dropped: 13 Cache: 0.0s

Exiting... (End of file)
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=7PIji8OubXU
 (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 --alang=eng (*) (opus 2ch 48000Hz)
File tags:
 Uploader: 8K Earth
 Channel_URL: https://www.youtube.com/channel/UChB3UnDddahXU7FKZXmpzMA
Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Could not dynamically load CUDA
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
[ffmpeg/video] vp9_rkmpp: An invalid frame was output by a decoder. This is a bug, please report it.
Error while decoding frame (hardware decoding)!
AO: [pipewire] 48000Hz stereo 2ch floatp
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
VO: [gpu] 3840x2160 drm_prime[p010]
AV: 00:12:15 / 00:12:16 (100%) A-V:  0.000 Dropped: 11 Cache: 0.0s

Exiting... (End of file)
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=KLuTLF3x9sA
 (+) Video --vid=1 (*) (vp9 3840x2160 30.000fps)
 (+) Audio --aid=1 --alang=eng (*) (opus 2ch 48000Hz)
File tags:
 Uploader: Relaxation Film
 Channel_URL: https://www.youtube.com/channel/UCPotnGNahFjLWjfsq4KYvuQ
Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Could not dynamically load CUDA
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
AO: [pipewire] 48000Hz stereo 2ch floatp
[ffmpeg/video] vp9_rkmpp: An invalid frame was output by a decoder. This is a bug, please report it.
Error while decoding frame (hardware decoding)!
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
VO: [gpu] 3840x2160 drm_prime[nv12]
AV: 01:28:00 / 11:59:13 (12%) A-V:  0.000 Dropped: 66 Cache: 111s/150MB

Exiting... (Quit)
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=k7dy1B6bOeM
 (+) Video --vid=1 (*) (vp9 3840x2160 59.940fps)
 (+) Audio --aid=1 --alang=eng (*) (eac3 6ch 48000Hz)
File tags:
 Uploader: 4K Video ULTRA HD
 Channel_URL: https://www.youtube.com/channel/UCRauo6y9tx0FB0OLLtqLA-g
Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Cannot load libcuda.so.1
[ffmpeg] AVHWDeviceContext: Could not dynamically load CUDA
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
AO: [pipewire] 48000Hz 5.1(side) 6ch floatp
VO: [gpu] 3840x2160 drm_prime[p010]
AV: 00:11:11 / 00:11:11 (100%) A-V:  0.000 Dropped: 16 Cache: 0.0s

Exiting... (End of file)
jfl@jfl-opi5plus:~$
```

Joshua-Riek, 2024-04-22:
Hmm well I guess that is some good news, but I'm not sure why it would be fixed if there was an issue. Either way we should keep an eye on this problem.

Joshua-Riek, 2024-04-22:
Yeah this looks to be the problem, its due to the `cloud-init` package conflicting with the user setup process.

JFLim1, 2024-04-23:
> @JFLim1 I may have found the issue, sometimes the desktop installer will not assign the default user to the `video` group. This is required to access the correct hardware interfaces for video playback.
```
jfl@jfl-opi5plus:~$ systemd-analyze blame
42.354s cloud-init-local.service
24.775s plymouth-quit-wait.service
 7.993s e2scrub_reap.service
 7.022s apparmor.service
```
```
jfl@jfl-opi5plus:~$ groups
jfl adm cdrom sudo dip video plugdev users lpadmin lxd sambashare
```
You mean cloud-init can interfere even after its service had been enabled?  Does normal Desktop user need "cloud-init"?  Cloud-init service is one of the contributing service that lengthen the boot up time.

Joshua-Riek, 2024-04-23:
I thought cloud-init was needed for the desktop as it's pulled in automatically as a dependency, but apparently, it is not needed. 

When the desktop installer creates the user it will not refresh the user groups on the first login. Requiring the user to log out and log back in for proper groups to be assigned. This is because cloud-init does some trickery during the boot process.

Joshua-Riek, 2024-04-27:
I'm going to close this now as unable to reproduce as it will not happen in the release Ubuntu 24.04 image.

closed 2024-04-27 by Joshua-Riek