Joshua-Riek/ubuntu-rockchip #721: ubuntu-server kernel 5.10 can not login by default user

ohyeah521, 2024-04-16:
i had build server 22.04 ker 5.10, but the default user ubuntu ,password ubuntu can not login.

ohyeah521, 2024-04-16:
I'll wait and see. I compile the firmware in sync with the latest code, unless the latest code affects the kernel 5.10 firmware.

Joshua-Riek, 2024-04-16:
If you compile from the source, there is a chance 22.04 is broken. I've been so focused on 24.04 and have ignored 22.04 for a couple of months.

After the 24.04 release, I plan to re-work 22.04 with a lot of improvements and fixes.

ProKn1fe, 2024-04-16:
Can confirm 22.04 default ubuntu/ubuntu not working.

Joshua-Riek, 2024-04-16:
Build from https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.33 for Ubuntu 22.04. I dont have time to fix it right now.

closed 2024-04-16 by Joshua-Riek

ohyeah521, 2024-04-17:
@Joshua-Riek Even after rolling back to version 1.33, it cannot be compiled. The error message is as follows:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/httplib2/__init__.py", line 1363, in _conn_request
    conn.connect()
  File "/usr/lib/python3/dist-packages/httplib2/__init__.py", line 1139, in connect
    address_info = socket.getaddrinfo(host, port, 0, socket.SOCK_STREAM)
  File "/usr/lib/python3.10/socket.py", line 955, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 364, in <module>
    sys.exit(0 if addaptrepo.main() else 1)
  File "/usr/bin/add-apt-repository", line 347, in main
    shortcut = handler(source, **shortcut_params)
  File "/usr/lib/python3/dist-packages/softwareproperties/shortcuts.py", line 40, in shortcut_handler
    return handler(shortcut, **kwargs)
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 82, in __init__
    if self.lpppa.publish_debug_symbols:
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 120, in lpppa
    self._lpppa = self.lpteam.getPPAByName(name=self.ppaname)
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 107, in lpteam
    self._lpteam = self.lp.people(self.teamname)
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 98, in lp
    self._lp = login_func("%s.%s" % (self.__module__, self.__class__.__name__),
  File "/usr/lib/python3/dist-packages/launchpadlib/launchpad.py", line 494, in login_anonymously
    return cls(
  File "/usr/lib/python3/dist-packages/launchpadlib/launchpad.py", line 230, in __init__
    super(Launchpad, self).__init__(
  File "/usr/lib/python3/dist-packages/lazr/restfulclient/resource.py", line 472, in __init__
    self._wadl = self._browser.get_wadl_application(self._root_uri)
  File "/usr/lib/python3/dist-packages/lazr/restfulclient/_browser.py", line 447, in get_wadl_application
    response, content = self._request(url, media_type=wadl_type)
  File "/usr/lib/python3/dist-packages/lazr/restfulclient/_browser.py", line 389, in _request
    response, content = self._request_and_retry(
  File "/usr/lib/python3/dist-packages/lazr/restfulclient/_browser.py", line 359, in _request_and_retry
    response, content = self._connection.request(
  File "/usr/lib/python3/dist-packages/httplib2/__init__.py", line 1693, in request
    (response, new_content) = self._request(
  File "/usr/lib/python3/dist-packages/launchpadlib/launchpad.py", line 144, in _request
    response, content = super(LaunchpadOAuthAwareHttp, self)._request(
  File "/usr/lib/python3/dist-packages/lazr/restfulclient/_browser.py", line 184, in _request
    return super(RestfulHttp, self)._request(
  File "/usr/lib/python3/dist-packages/httplib2/__init__.py", line 1441, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "/usr/lib/python3/dist-packages/httplib2/__init__.py", line 1370, in _conn_request
    raise ServerNotFoundError("Unable to find the server at %s" % conn.host)
httplib2.error.ServerNotFoundError: Unable to find the server at api.launchpad.net
Error: in ./scripts/config-image.sh on line 88
Error: in ./build.sh on line 221

```

Joshua-Riek, 2024-04-17:
Its likely that the ubuntu servers are down for the moment. Try in an hour.

ohyeah521, 2024-04-17:
thank you , i'll try latter

ohyeah521, 2024-04-17:
@ProKn1fe  Hello, can you roll back to version 1.33? I had a compilation error after rolling back. Is that normal for you?

ohyeah521, 2024-04-17:
this will help:https://github.com/Joshua-Riek/ubuntu-rockchip/commit/64ea4f286fb7fc141a68ed6829ad45f547c82f74

netPokey, 2024-04-23:
@ohyeah521 你好 我与你遇到同样的问题 我自己编译的  ubuntu 密码错误 我觉得 cloud-init  没起作用 但是我无法修复 您在解决的

netPokey, 2024-04-23:
> 所有者

如果我无法连接ubuntu 服务器该怎么办呢

ohyeah521, 2024-04-23:
@netPokey 回退到1.33版本

netPokey, 2024-04-23:
> @netPokey回退到1.33版本

谢谢 ，我看到您有添加过ebpf的部分 但是我现在去找你补丁部分和编译的包都没有找到 可以提供借鉴一下吗，我在考虑吧ebpf和 npu0.9.6驱动pacth

ohyeah521, 2024-04-23:
没成功，后来换armbian玩了一下

netPokey, 2024-04-23:
我看了iuess 有个大佬有提供默认密码和网络的的修改patch

ohyeah521, 2024-04-23:
你可以试试，暂时不想搞了，等过了五一再说吧