Joshua-Riek/ubuntu-rockchip #722: Radxa Zero 3W stuck in boot loop 24.04-beta-preinstalled-desktop-arm64-radxa-zero3.img

DrKnow65, 2024-04-16:
Radxa Zero 3 W boots well on SD to 24.04-3.
When I try to run 24.04-beta I get an initial error of;
[FAILED] Failed to staet apparmor.service - Load AppArmor profiles

It then goes through some SSH key generation stuff and eventually lands on a repeating;

rk_gmac-dwmac fe10000.ethernet: Failed to reset the dma
rk_gmac-dwmac fe10000.ethernet end1: stmmac_hw_setup: DMA engine initialization failed

I have tried using etcher on Windows and Linux boxes, dding the .img from the SD to EMMC while  booted into 24.04-3.

Is there any info I can look for to help diagnose the issue?

Joshua-Riek, 2024-04-16:
How much RAM does your board have?

DrKnow65, 2024-04-16:
I have the 8Gb/64GB EMMC variant.

Joshua-Riek, 2024-04-24:
I tested the latest build on GitHub actions and can not reproduce this problem.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/8808046061

closed 2024-04-24 by Joshua-Riek