Joshua-Riek/ubuntu-rockchip #723: Orange Pi 3b (RK3566) 24.04 beta, no desktop after normal installation

1chard, 2024-04-16:
Using desktop image with installer fix, i was able to configure user and keyboard like as expected, but after installer was completed, the desktop not loaded at all, the shell showed some error on apparmor service load, some ssh code, then started to loop into a black screen and some code. I manually restarted using ssh and after reboot, plymouth ran normally, but after gdm3 service load, it was a single black screen.

The same behavior was reported for [this opi 3b discussion](https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#discussioncomment-9124542) and this radxa 3w issue (also rk3566) #722.

I also tried to manually run gnome-shell manually using ´XDG_SESSION_TYPE=wayland dbus-run-session gnome-shell´, but it did not work. 

I attached some logs of gdm3 status, ubiquiti "/var/log/installer/dm" log and manual gnome-shell run below.

[gdm service log.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15002016/gdm.service.log.txt)
[ubiquiti 2.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15002029/ubiquiti.2.txt)
[gnome manual run.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15002036/gnome.manual.run.txt)

Joshua-Riek, 2024-04-16:
How much RAM does your board have?

1chard, 2024-04-16:
It is the 4gb model

edsmat95, 2024-04-20:
Same issue

Joshua-Riek, 2024-04-21:
I will look into it, but I need time to finish other improvements.

Joshua-Riek, 2024-04-24:
I had some time to take a look at this and I will not fix this issue for a while and likely remove the desktop image for now. I'm so tired and just want to get something stable out for RK3588, then I need a break.

lukaszsobala, 2024-04-27:
@1chard I just had the same error. First thing I would try is to remove the panfork ppa as rk3566 shouldn't work with it, but mainline mesa. I will do this and report back.

Joshua-Riek, 2024-04-27:
Yeah that sounds like a good start.

lukaszsobala, 2024-04-27:
Woo, It works!

I'll send the commands later. Initramfs generation finished with "unsupported platform"  message but it worked anyway.

lukaszsobala, 2024-04-27:
Hi!

so the solution is to:

0. Install ubuntu 24.04, reboot, wait for gdm to fail, connect via `ssh`
1. Use `sudoedit` to comment out the two panfork ppa lines in `/etc/apt/sources.list.d/extra-ppas.list` (ppa-purge does not seem to work)
2. `sudo apt remove libgl1-mesa-dri mesa-va-drivers mesa-vulkan-drivers libgbm1 libglapi-mesa libegl-mesa0` (this will remove a lot of packages)
3. `sudo apt install ubuntu-desktop-minimal` or not minimal if one prefers
4. `sudo apt upgrade -y && sudo apt full-upgrade -y --auto-remove` (I did it at the beginning though`
5. reboot, enjoy the desktop experience

Joshua-Riek, 2024-04-27:
Flash kernel should not be installed, extlinux is now used

Joshua-Riek, 2024-04-27:
So it's safe to purge flash kernel

lukaszsobala, 2024-04-27:
Purged, there are no errors now.

lukaszsobala, 2024-04-28:
I also tried first commenting out the ppa after flashing the image, then installing the Ubuntu. It did not change anything - the preinstalled packages are still from the ppa, so the rest should be done still.

Thanks for making the distribution!

1chard, 2024-04-28:
Nice catch @lukaszsobala ! gnome started as expected, on wayland, but at least for me, hw video decoder is missing (tested on moonlight-qt) and this worked on Ubuntu 22.04 

any idea?

lukaszsobala, 2024-04-28:
> hw video decoder is missing (tested on moonlight-qt) and this worked on Ubuntu 22.04 
> 
> any idea?

Hmm, I'm not an expert on this at all, sorry. Which formats? In wayland Kodi, DRM PRIME works for me up to 1080p, I tried one 4k video and the screen was black. Youtube seems to work in Firefox but not Chromium.

Joshua-Riek, 2024-04-28:
Should be fixed by removing panfork with this commit https://github.com/Joshua-Riek/ubuntu-rockchip/commit/ac9a8fd94b0e9a4c3cbddcbabd43bc218802679f.

Joshua-Riek, 2024-05-05:
I've been re-working things and rk3566 should be good for the next release.

However Idk if ffmpeg, gstreamer, etc will work. If anyone has insight on RK3566 in this area please let me know,

edsmat95, 2024-05-05:
> Oi! Escrevendo isso no Ubuntu 24.04 no OrangePi 3B!
> 
> então a solução é:
> 
> 0. Instale o Ubuntu 24.04, reinicie, espere o gdm falhar, conecte-se via`ssh`
> 1. Use `sudoedit`para comentar as duas linhas panfork ppa `/etc/apt/sources.list.d/extra-ppas.list`(ppa-purge não parece funcionar)
> 2. `sudo apt remove libgl1-mesa-dri mesa-va-drivers mesa-vulkan-drivers libgbm1 libglapi-mesa libegl-mesa0`(isso removerá muitos pacotes)
> 3. `sudo apt install ubuntu-desktop-minimal`ou não mínimo se preferir
> 4. `sudo apt upgrade -y && sudo apt full-upgrade -y --auto-remove`(mas eu fiz isso no começo)
> 5. reinicie, aproveite a experiência da área de trabalho
> 
> Mas há um problema, acho que no kernel mais recente@Joshua-Riekremoveu o suporte da placa e este é o resultado do apt upgrade:
> 
> ```
> Unsupported platform 'Rockchip RK3566 OPi 3B'.
> dpkg: error processing package flash-kernel (--configure):
>  installed flash-kernel package post-installation script subprocess returned err
> or exit status 1
> Processing triggers for initramfs-tools (0.142ubuntu25) ...
> update-initramfs: Generating /boot/initrd.img-6.1.0-1011-rockchip
> Unsupported platform 'Rockchip RK3566 OPi 3B'.
> run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
> dpkg: error processing package initramfs-tools (--configure):
>  installed initramfs-tools package post-installation script subprocess returned 
> error exit status 1
> Errors were encountered while processing:
>  flash-kernel
>  initramfs-tools
> ```
> 
> Você poderia reativar o suporte?

To work is it a build server or desktop?

lukaszsobala, 2024-05-06:
> If anyone has insight on RK3566 in this area please let me know

I tried jellyfin from nyanmisaka (https://hub.docker.com/r/nyanmisaka/jellyfin), enabled RKMPP transcoding (with or without HEVC) and it does not seem to transcode (I get "Source error"). Native play works fine. Maybe there is some config magic that can be done to fix it.

nyanmisaka, 2024-05-07:
Due to hardware constraints, the rk356x can only transcode to 1080p and lower. And it does not support HDR2SDR tone-mapping. See also: https://jellyfin.org/docs/general/administration/hardware-acceleration/rockchip/

lukaszsobala, 2024-05-07:
@nyanmisaka if it did that it would be great, but now, for some reason, it does not, at least using jellyfin.

Could you send an ffmpeg command example so that I can see where it fails?

nyanmisaka, 2024-05-07:
> Could you send an ffmpeg command example so that I can see where it fails?

Test commands can be found in https://github.com/nyanmisaka/ffmpeg-rockchip/wiki

lukaszsobala, 2024-05-09:
> > Could you send an ffmpeg command example so that I can see where it fails?
> 
> Test commands can be found in https://github.com/nyanmisaka/ffmpeg-rockchip/wiki

I got this when trying to transcode using the command given in the transcode (hw/hw) example - modified to 720p (1080p input):

`Parsed_scale_rkrga_0 @ 0xaaaab04760f0] Output format 'nv12' with AFBC modifier is not supported by RGA2`

The transcode worked at above 4x speed but I'm not sure if it used hardware.

nyanmisaka, 2024-05-09:
> I got this when trying to transcode using the command given in the transcode (hw/hw) example - modified to 720p (1080p input):
>
> `Parsed_scale_rkrga_0 @ 0xaaaab04760f0] Output format 'nv12' with AFBC [...]

It's just a warning, just remove `:afbc=1`. 4x realtime speed is definitely hardware encoding.

lukaszsobala, 2024-05-09:
I thought so. Now how to find out why it doesn't work in jellyfin...?

nyanmisaka, 2024-05-09:
RK356x cannot handle HDR, so HDR tone-mapping needs to be disabled.
Also it doesn't support video encoding larger than 1080p, you need to limit it manually.
Or you are not using privileged mode to pass multiple device files to docker.

lukaszsobala, 2024-05-09:
All of these options are already set up this way.

Joshua-Riek, 2024-05-11:
This has been fixed and should be in the next release. I hope to have ready tonight or tomorrow

closed 2024-05-11 by Joshua-Riek