Joshua-Riek/ubuntu-rockchip #724: Inbuilt video player issues Ubuntu 24.04 LTS Beta v2.0.0-beta Orange Pi 5

Bluesmanuk, 2024-04-18:
When attempting to play local videos of many types and variety, when using the built in video player, most of them would be result in a grey screen with what appears to be a broken camera icon.

One or two resulted in an error message 'The playback of this movie requires MPEG-1 system screen demuxer plugin which is not installed'.

I  installed VLC player but whilst some videos would at least begin to play (many had black screens), initial smooth playback would soon stall or stutter after a few seconds, with some then generating massively pixelated images.

Kodi fared a little better but when turning on the option for hardware accelerated video, quite a few videos would simply show a black screen.

Joshua-Riek, 2024-04-18:
VLC is not supported. What video player are you talking about? MPV should be working perfectly.

Joshua-Riek, 2024-04-18:
Totem is not supported and can not use hardware acceleration. The default media player should be mpv.

bagong, 2024-04-18:
Is there any priority which mediaplayers to support? I'd love to see VLC and ffmpeg working at some point. If it's too much work, we could maybe try to aggregate generic multimedia compile instructions to make it easier...

Joshua-Riek, 2024-04-18:
ffmpeg and MPV work great, but it's not my interest or specialty. There are a few great devs I know who are very knowledgeable in this area, I can reach out. But I need time to finish the build system changes, so it is easier to maintain in the future.

Bluesmanuk, 2024-04-18:
MPV it is then. I guess that Totem could be removed from the build.

closed 2024-04-18 by Bluesmanuk