Joshua-Riek/ubuntu-rockchip #726: 24.04 BETA ROCK5A - Logitech MX Master 3 Mouse Doesnt Work

MyVizDrake, 2024-04-19:
Rock 5A and 6.1 Kernel. USB Dongle based Logitech MX Master 3 Mouse doesn't work at all. Logitech M510 works fine.

I can provide whatever additional information is requested.  Just let me know!

MyVizDrake, 2024-04-19:
No issues!  Just documenting issues as requested.  Appreciate your efforts and look forward to the day when I can use this on my Rock5A !

Xeue, 2024-04-20:
I saw this initally, but at some stage my one started working...

MyVizDrake, 2024-04-27:
@Xeue yeah, no luck for me ... still not working.  I'll wait patiently and see if we can resolve the A8 Bluetooth issue in which case I will use a Bluetooth Mouse / Keyboard.

Joshua-Riek, 2024-04-30:
Should work with the noble release, please re-open if still an issue.

closed 2024-04-30 by Joshua-Riek

MyVizDrake, 2024-05-02:
I had to change the channel on the mouse itself for it to work but it is in fact working.  Moving on! :-)