Joshua-Riek/ubuntu-rockchip #727: 24.04 BETA ROCK5A - Radxa A8 Module Issues

MyVizDrake, 2024-04-19:
Rock 5A w/ Radxa Wireless A8 and Kernel 6.1

WLAN seems to be OK (can nslookup or dig a domain) but cannot ping. Error says 

ping: socket: operation not permitted
ping: => missing cap_net_raw+p capability or setuid

Bluetooth is disabled and cannot be enabled so I cannot connect my alternate keyboard / mouse..

while testing WLAN via Software Update, it was checking / downloading ports and then the entire system locked up.

Can provide whatever additional information / logs as requested.

Joshua-Riek, 2024-04-21:
Can you put the output of dmesg into a text file and upload it here? Also, what is the output of the command `ip a`?

MyVizDrake, 2024-04-22:
As requested
[dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15065310/dmesg.txt)
[ip_a.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15065311/ip_a.txt)

Joshua-Riek, 2024-04-26:
Ok so, I found a work around, on the command line enter the below commands.
```
sudo modprobe -r btusb
sudo modprobe btusb
```
Running these commands should not be required for bluetooth to work properly, but this is a start at least.

MyVizDrake, 2024-04-26:
I did a quick test and it didn’t work but I am not convinced that I wasn’t the cause. Will revisit this in the next day or so …
 Appreciate it! 

 Issues (Issue #727) as subject

MyVizDrake, 2024-04-27:
Tried it again and it definitely did not work  Made more difficult as I have no available mouse since my Logitech MX MAaster 3 doesn't work either.  Just kicked off apt upgrade in the hope that maybe something will change. I doubt it will work, but .... :-)

Let me know if I need to get anything else off of here to troubleshoot!

Joshua-Riek, 2024-04-30:
Should work with the noble release, please re-open if still an issue.

closed 2024-04-30 by Joshua-Riek

Joshua-Riek, 2024-05-02:
Try to enter the command `sudo modprobe btusb` and it should work.

MyVizDrake, 2024-05-02:
I tried sudo modprobe -r btusb .... sudo modprobe btusb and just tried sudo modprobe btusb again at your direction .... nothing is showing up in the Gnome Bluetooh settings window.  It's not that it can't see the two devices I care about, it also can't see all of the other USB devices that Android image could see.  That being said, my ArchLinux install that actually has a working DSI display can't see any BT devices either ...

MyVizDrake, 2024-05-02:
The BT device is available and running from what I can see ..>

[image]