Joshua-Riek/ubuntu-rockchip #728: Cannot install libgl1-mesa-dri:armhf on Ubuntu 24.04

lukaszsobala, 2024-04-19:
This package depends on `libllvm14` which is unavailable on Ubuntu Noble.

The closest available package in `libllvm14t64`. I could change the name in the control file of the deb package and it installed correctly.

Joshua-Riek, 2024-04-20:
Some libs where renamed due to the 32-bit time problem. Let me take a look at the dependencies of panfork and see what I can do.

Joshua-Riek, 2024-04-20:
Ah I see the problem, I must re-build package to pull in the dependency name change.

lukaszsobala, 2024-04-20:
> Ah I see the problem, I must re-build package to pull in the dependency name change.

Yes, the dependency name changed. Thanks!

Joshua-Riek, 2024-04-20:
Hmm waiting for dependeny to update, it may take a few days for the Launchpad server to sync.

https://launchpad.net/~jjriek/+archive/ubuntu/panfork-mesa/+build/28129053

lukaszsobala, 2024-04-26:
6 days and counting, is there a way to check how long to go?

Joshua-Riek, 2024-04-26:
I've been trying to fix it for a few days with no luck. I'm going to try a hack next to fix things. The armhf package somehow finds libllvm14 instead of libllvm14t64 and libllvm14 no longer exists. 

I will try manually adding the dependency for libgl1-mesa-dri as shown below. 
```
Package: libgl1-mesa-dri
Section: libs
Architecture: any
Depends:
 libc6,
 libdrm-nouveau2,
 libdrm-radeon1,
 libdrm2,
 libexpat1,
 libgcc-s1,
 libglapi-mesa,
 libllvm14t64,
 libsensors5,
 libstdc++6,
 libxcb-dri3-0,
 libzstd1,
 zlib1g,
 mali-g610-firmware
```

lukaszsobala, 2024-04-26:
It seems that is basically what I did (but using the "control" file in DEBIAN in the uncompressed deb package)

Joshua-Riek, 2024-04-26:
Did it compile on your end?

lukaszsobala, 2024-04-26:
I did not compile it at all but did more or less this: https://forums.linuxmint.com/viewtopic.php?t=35136

Joshua-Riek, 2024-04-27:
I was able to get the package to build on a testing ppa for Launchpad. I will upload to my main ppa and should be available in a few hours.

Joshua-Riek, 2024-04-27:
Its now on Launchpad and should work:

https://launchpad.net/~jjriek/+archive/ubuntu/panfork-mesa/+packages

closed 2024-04-27 by Joshua-Riek