Joshua-Riek/ubuntu-rockchip #730: sysctl: cannot stat /proc/sys/kernel/apparmor_restrict_unprivileged_userns

Al3XKOoL, 2024-04-20:
Hi. apparmor not work.

`root@opi-usb:/home/user# /etc/init.d/apparmor restart
Restarting apparmor (via systemctl): apparmor.serviceJob for apparmor.service failed because the control process exited with error code.
See "systemctl status apparmor.service" and "journalctl -xeu apparmor.service" for details.
 failed!
root@opi-usb:/home/user# journalctl -xeu apparmor.service
░░ The unit apparmor.service has entered the 'failed' state with result 'exit-code'.
Apr 20 10:26:57 opi-usb systemd[1]: Failed to start apparmor.service - Load AppArmor profiles.
░░ Subject: A start job for unit apparmor.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ A start job for unit apparmor.service has finished with a failure.
░░
░░ The job identifier is 53 and the job result is failed.
Apr 20 10:26:57 opi-usb systemd[1]: apparmor.service: Consumed 9.671s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ The unit apparmor.service completed and consumed the indicated resources.
Apr 20 12:08:29 opi-usb systemd[1]: Starting apparmor.service - Load AppArmor profiles...
░░ Subject: A start job for unit apparmor.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ A start job for unit apparmor.service has begun execution.
░░
░░ The job identifier is 2536.
Apr 20 12:08:29 opi-usb apparmor.systemd[2109]: Restarting AppArmor
Apr 20 12:08:29 opi-usb apparmor.systemd[2113]: sysctl: cannot stat /proc/sys/kernel/apparmor_restrict_unprivileged_userns: No such file or directory
Apr 20 12:08:29 opi-usb apparmor.systemd[2109]: Reloading AppArmor profiles
Apr 20 12:08:30 opi-usb apparmor.systemd[2203]: /sbin/apparmor_parser: Unable to replace "unprivileged_userns".  Profile doesn't conform to protocol
Apr 20 12:08:30 opi-usb apparmor.systemd[2163]: /sbin/apparmor_parser: Unable to replace "QtWebEngineProcess".  Profile doesn't exist
Apr 20 12:08:30 opi-usb apparmor.systemd[2163]: /sbin/apparmor_parser: Unable to replace "plasmashell".  Profile doesn't conform to protocol
Apr 20 12:08:30 opi-usb apparmor.systemd[2216]: Warning: found usr.sbin.sssd in /etc/apparmor.d/force-complain, forcing complain mode
Apr 20 12:08:30 opi-usb apparmor.systemd[2158]: /sbin/apparmor_parser: Unable to replace "kmod".  Profile doesn't exist
Apr 20 12:08:30 opi-usb apparmor.systemd[2158]: /sbin/apparmor_parser: Unable to replace "nvidia_modprobe".  Profile doesn't conform to protocol
Apr 20 12:08:30 opi-usb apparmor.systemd[2216]: Warning from /etc/apparmor.d (/etc/apparmor.d/usr.sbin.sssd line 63): Caching disabled for: 'usr.sbin.sssd' due to force complain
Apr 20 12:08:30 opi-usb apparmor.systemd[2206]: /sbin/apparmor_parser: Unable to replace "/usr/bin/man".  Profile doesn't conform to protocol
Apr 20 12:08:30 opi-usb apparmor.systemd[2214]: /sbin/apparmor_parser: Unable to replace "third_party".  Profile doesn't exist
Apr 20 12:08:30 opi-usb apparmor.systemd[2214]: /sbin/apparmor_parser: Unable to replace "/usr/sbin/cupsd".  Profile doesn't conform to protocol
Apr 20 12:08:32 opi-usb apparmor.systemd[2205]: /sbin/apparmor_parser: Unable to replace "sanitized_helper".  Profile doesn't exist
Apr 20 12:08:32 opi-usb apparmor.systemd[2205]: /sbin/apparmor_parser: Unable to replace "/usr/bin/evince".  Profile doesn't conform to protocol
Apr 20 12:08:36 opi-usb apparmor.systemd[2210]: /sbin/apparmor_parser: Unable to replace "gpg".  Profile doesn't exist
Apr 20 12:08:36 opi-usb apparmor.systemd[2210]: /sbin/apparmor_parser: Unable to replace "libreoffice-soffice".  Profile doesn't conform to protocol
Apr 20 12:08:36 opi-usb apparmor.systemd[2109]: Error: At least one profile failed to load
Apr 20 12:08:36 opi-usb systemd[1]: apparmor.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ An ExecStart= process belonging to unit apparmor.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Apr 20 12:08:36 opi-usb systemd[1]: apparmor.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ The unit apparmor.service has entered the 'failed' state with result 'exit-code'.
Apr 20 12:08:36 opi-usb systemd[1]: Failed to start apparmor.service - Load AppArmor profiles.
░░ Subject: A start job for unit apparmor.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ A start job for unit apparmor.service has finished with a failure.
░░
░░ The job identifier is 2536 and the job result is failed.
Apr 20 12:08:36 opi-usb systemd[1]: apparmor.service: Consumed 9.376s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ The unit apparmor.service completed and consumed the indicated resources.`

Al3XKOoL, 2024-04-22:
It's Ubuntu 24.04 beta.

Apparmor failed to start profiles with:
-Profile doesn't conform to protocol
-Profile doesn't exist.
 Also "/proc/sys/kernel/apparmor_restrict_unprivileged_userns" is not enabled?

Joshua-Riek, 2024-04-22:
I'm not very familiar with Apparmor, I was able to add support in the kernel but I guess it needs some configuration. Would you happen to have any insight in this area?

Al3XKOoL, 2024-04-22:
Not to much but i can find some info.

Comparing Ubuntu 22.04.3 to Ubuntu 23.10 introduces 5 new apparmor related entities in /proc/sys/kernel/: apparmor_restrict_unprivileged_unconfined, apparmor_restrict_unprivileged_userns_complain, apparmor_restrict_unprivileged_io_uring, apparmor_restrict_unprivileged_userns, apparmor_restrict_unprivileged_userns_force

I think these are needed enabled in kernel.

Maybe this explain a little:
https://ubuntu.com/blog/ubuntu-23-10-restricted-unprivileged-user-namespaces

Joshua-Riek, 2024-04-22:
Thanks, I will take a look, if I can't figure it out in a few hours I will need to come back to this at a later date and disable apparmor for the moment.

Al3XKOoL, 2024-04-22:
Ok no problem, you are doing good work.  Take your time.

I fixed apparmor by disabling all not working profiles for now, but i thinks something is wrong. Docker frigate is not working well with your beta version.

[image]

Joshua-Riek, 2024-04-27:
Yeah I'm disabling Apparmor. It's to much of a mess with the vendor rockchip 6.1 kernel.

closed 2024-04-27 by Joshua-Riek