Joshua-Riek/ubuntu-rockchip #735: Docker Desktop not starting after install

wgrs, 2024-04-21:
I installed Docker Desktop. The install seemed to run correctly but Docker Desktop wont start.

I completely removed Docker and reinstalled, still no success.

Joshua-Riek, 2024-04-21:
I've been able to user docker just fine from command line, but my guess is there is a problem with the desktop application.

wgrs, 2024-04-22:
> I've been able to user docker just fine from command line, but my guess is there is a problem with the desktop application.

Can you give the install procedure you used?

alessandroRoaro, 2024-04-22:
I am also having this issue. I also noticed that if I try to run for example "docker ps -a" then the command gets stuck and nothing happens

wgrs, 2024-04-22:
> I installed docker from command line with `sudo apt-get install docker.io` and it works for me. How did you both install docker exactly?

Same. My issue is that I need Docker Desktop running. The daemon won't start.

othavioquiliao, 2024-04-23:
I'm also having the same problem, the Docker Desktop (downloaded the .deb file directly from Docker) simply won't open, but it's working normally via terminal.

wgrs, 2024-04-23:
> Does anyone have logs of this behavior? I do not have time right now to test this out, but I can look through the logs if provided.

How do I get logs?

Freezystem, 2024-04-26:
> I am also having this issue. I also noticed that if I try to run for example "docker ps -a" then the command gets stuck and nothing happens

I fixed it by running: `docker context use default`
It is explained here in [this guide](https://phoenixnap.com/kb/cannot-connect-to-the-docker-daemon-error).
Unfortunately it's fixing docker but it's not fixing docker desktop startup.

bonezee, 2024-04-26:
> I fixed it by running: `docker context use default` It is explained here in [this guide](https://phoenixnap.com/kb/cannot-connect-to-the-docker-daemon-error). Unfortunately it's fixing docker but [...]

Thank you -  this guide helped me get docker via terminal working but still no joy with Docker Desktop,   I'm waiting patiently for this bug to me fixed.

nathannli, 2024-04-26:
@Joshua-Riek some logs:
launching docker desktop via
`systemctl --user start docker-desktop`

then following the logs via `journalctl --user -u docker-desktop.service -f`
```
Apr 26 16:57:22 ubuntu7950x systemd[1731]: Stopped docker-desktop.service - Docker Desktop.
Apr 26 16:57:30 ubuntu7950x systemd[1731]: Starting docker-desktop.service - Docker Desktop...
Apr 26 16:57:30 ubuntu7950x systemd[1731]: Started docker-desktop.service - Docker Desktop.
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.610641265Z][com.docker.backend.main][W] system telemetry disabled, custom otlp endpoint "https://desktop.observability-stage.docker.com" will be ignored
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.610843759Z][com.docker.backend.sys][I] maximum number of file descriptors is 1048576
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.610873228Z][com.docker.backend.compatibility][I] Virtualization enabled
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.610916067Z][com.docker.backend.migrations][I] running migrations
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.612917638Z][com.docker.backend.settingsloader][I] loaded settings
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.613009205Z][com.docker.backend][I] host will use proxy:  http_proxy=none https_proxy=none tcp_proxy=none noproxy=
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.613028754Z][com.docker.backend][I] Linux will use proxy:  http_proxy=none https_proxy=none tcp_proxy=none noproxy=
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.613034794Z][com.docker.backend][I] host will use proxy:  http_proxy=none https_proxy=none tcp_proxy=none noproxy=
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.613039754Z][com.docker.backend][I] Linux will use proxy:  http_proxy=none https_proxy=none tcp_proxy=none noproxy=
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614500301Z][com.docker.backend.state][I] sending desktop state:InitState
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614704865Z][com.docker.backend.component][I] registring publisher:{kubernetes}
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614717245Z][com.docker.backend.ipc][I] 54839609-kubernetes -> <HOME>/.docker/desktop/lifecycle-server.sock VMDockerdAPI
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614729224Z][com.docker.backend.kubernetes][I] broadcasting Kubernetes event: [2024-04-26T16:57:30.610559357-04:00] (cffa9928) kubernetes disabled : Kubernetes is stopped
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614737514Z][com.docker.backend.tracker][I] sending event: desktopState
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614782733Z][com.docker.backend.ipc][I] fb95deba-PauseHDL -> /run/guest-services/procd.sock procd
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614794862Z][com.docker.backend.ipc][I] 5cfed482-registryaccessPKG -> <HOME>/.docker/desktop/backend.sock BackendAPI
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614872740Z][com.docker.backend.ipc][I] 218a85e5-engines -> <HOME>/.docker/desktop/lifecycle-server.sock VMDockerdAPI
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614884710Z][com.docker.backend.ipc][I] 7a91f898-serviceClientCLI ->  com.docker.service
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614911479Z][com.docker.backend.idle][I] initializing idle monitor
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614924089Z][com.docker.backend.ipc][I] 26433a47-idle -> /run/guest-services/lifecycle-server.sock lifecycle
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614932448Z][com.docker.backend.ipc][I] d6691b6b-idle -> <HOME>/.docker/desktop/virtualization.sock balloon
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614974897Z][com.docker.backend.ipc][I] 3890e38e-devenvs-heartbeat -> <HOME>/.docker/desktop/docker-dev-env-api.sock DevEnvsServer
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.614983457Z][com.docker.backend.ipc][I] 98514b24-volume-contents -> <HOME>/.docker/desktop/volume-contents.sock DockerVolumeServer
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.615073154Z][com.docker.backend.ipc][I] 5df2d879-DockerApiProxyBackendClient -> <HOME>/.docker/desktop/backend.sock BackendAPI
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.615213920Z][com.docker.backend.ipc][I] 08a3ca49-DockerApiProxyVolumeApprover -> <HOME>/.docker/desktop/backend.sock BackendAPI
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.615263179Z][com.docker.backend.ipc][I] 96205dc8-DockerApiProxyVolumeSharer -> <HOME>/.docker/desktop/backend.sock BackendAPI
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.615269368Z][com.docker.backend.apiproxy][I] reinitializing: creating new proxies
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.615277148Z][com.docker.backend.apiproxy][I] creating Docker API proxy for Linux engine
[... cut ...]
Apr 26 16:57:35 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:35.893629253Z][com.docker.backend.segment][I] response 200 OK
```

install method:
1) download .deb file from https://docs.docker.com/desktop/install/ubuntu/
2) add docker ppa from https://docs.docker.com/engine/install/ubuntu/#install-using-the-repository
3) sudo apt install ./{deb file name}.deb
4) {for my system, i also did this} follow instructions to rm sudo from https://docs.docker.com/engine/install/linux-postinstall/

Joshua-Riek, 2024-04-26:
Looks like an internal problem with Docker desktop, I do not think this is something I can fix.
```
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [2024-04-26T20:57:30.639339108Z][com.docker.backend.tasks][E] supervisor detected that task has died with an error: shutdown with exit code 1: docker
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [common/cmd/com.docker.backend/supervisor/tasks.monitor({0x31a35a8, 0xc000600370}, {0xc0001497b0?, 0x2?}, 0xc00074d5f0)
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [        common/cmd/com.docker.backend/supervisor/tasks/supervisor.go:73 +0xf3
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [common/cmd/com.docker.backend/supervisor/tasks.Supervise.func2()
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [        common/cmd/com.docker.backend/supervisor/tasks/supervisor.go:47 +0xce
Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [common/cmd/com.docker.backend/supervisor/tasks.Supervise in goroutine 245
```

wgrs, 2024-04-26:
> I've been able to user docker just fine from command line, but my guess is there is a problem with the desktop application.

I am testing from the command line. The docker daemon won't start.

wgrs, 2024-04-26:
> Looks like an internal problem with Docker desktop, I do not think this is something I can fix.
>
> ```
> Apr 26 16:57:30 ubuntu7950x com.docker.backend[16472]: [...]

Thanks for checking

k3bra, 2024-04-27:
I'm having the same issue, also when I try to build the project on the terminal it just hangs there and nothing happen.

closed 2025-02-17 by github-actions[bot]