Joshua-Riek/ubuntu-rockchip #738: Cameras not working on Ubuntu 24.04 Beta

peytonicmaster6, 2024-04-23:
I'm having an issue getting basic camera functionality to work on Ubuntu 24.04 Desktop, with no success of pulling up any sort of camera feed. I have an OV13855 plugged into an Orange Pi 5B on camera slot 1, with the overlay set like so:
`U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-ov13855-c1.dtbo"`

The output of dmesg looks like the following:
> [   11.097850] platform csi2-dphy0: Fixed dependency cycle(s) with /i2c@fec90000/ov13855-1@36
> [   11.281087] ov13855 7-0036: driver version: 00.01.04
> [   11.281124] ov13855 7-0036: Failed to get power-gpios, maybe no use
> [   11.281166] ov13855 7-0036: Looking up avdd-supply from device tree
> [   11.281175] ov13855 7-0036: Looking up avdd-supply property in node /i2c@fec90000/ov13855-1@36 failed
> [   11.281208] ov13855 7-0036: supply avdd not found, using dummy regulator
> [   11.281290] ov13855 7-0036: Looking up dovdd-supply from device tree
> [   11.281299] ov13855 7-0036: Looking up dovdd-supply property in node /i2c@fec90000/ov13855-1@36 failed
> [   11.281320] ov13855 7-0036: supply dovdd not found, using dummy regulator
> [   11.281352] ov13855 7-0036: Looking up dvdd-supply from device tree
> [   11.281361] ov13855 7-0036: Looking up dvdd-supply property in node /i2c@fec90000/ov13855-1@36 failed
> [   11.281380] ov13855 7-0036: supply dvdd not found, using dummy regulator
> [   11.281411] ov13855 7-0036: could not get default pinstate
> [   11.281473] ov13855 7-0036: could not get sleep pinstate
> [   11.291828] ov13855 7-0036: Detected OV00d855 sensor, REVISION 0xb0
> [   27.175263] ov13855 7-0036: Consider updating driver ov13855 to match on endpoints
> [   27.175267] rockchip-csi2-dphy csi2-dphy0: dphy0 matches m00_b_ov13855 7-0036:bus type 5

For kernal 5.10, dmesg looks like the following.
> [    4.937666] ov13855 7-0036: driver version: 00.01.04
> [    4.937725] ov13855 7-0036: Failed to get power-gpios, maybe no use
> [    4.937840] ov13855 7-0036: Looking up avdd-supply from device tree
> [    4.937857] ov13855 7-0036: Looking up avdd-supply property in node /i2c@fec90000/ov13855-1@36 failed
> [    4.937893] ov13855 7-0036: supply avdd not found, using dummy regulator
> [    4.938081] ov13855 7-0036: Looking up dovdd-supply from device tree
> [    4.938097] ov13855 7-0036: Looking up dovdd-supply property in node /i2c@fec90000/ov13855-1@36 failed
> [    4.938117] ov13855 7-0036: supply dovdd not found, using dummy regulator
> [    4.938171] ov13855 7-0036: Looking up dvdd-supply from device tree
> [    4.938186] ov13855 7-0036: Looking up dvdd-supply property in node /i2c@fec90000/ov13855-1@36 failed
> [    4.938203] ov13855 7-0036: supply dvdd not found, using dummy regulator
> [    4.938311] ov13855 7-0036: could not get default pinstate
> [    4.938453] ov13855 7-0036: could not get sleep pinstate
> [    4.949136] ov13855 7-0036: Detected OV00d855 sensor, REVISION 0xb0
> [    5.021811] rockchip-csi2-dphy csi2-dphy0: dphy0 matches m00_b_ov13855 7-0036:bus type 5

and i2cdetect is showing a device on 0x36, the address of the sensor for both 5.10 and 6.1.
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- UU -- -- -- 
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- 
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 50: -- -- -- -- -- -- -- -- 58 -- -- -- -- -- -- -- 
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 70: -- -- -- -- -- -- -- --   

Running v4l2-ctl --list-ctrls on the correct /dev/v4l-subdevX device provides the following output.
> User Controls
> 
>         white_balance_automatic 0x0098090c (bool)   : default=0 value=0
>                        exposure 0x00980911 (int)    : min=4 max=1964 step=1 default=1000 value=1181
>                  gain_automatic 0x00980912 (bool)   : default=0 value=0
>                 horizontal_flip 0x00980914 (bool)   : default=0 value=0 flags=modify-layout
>                   vertical_flip 0x00980915 (bool)   : default=0 value=0 flags=modify-layout
> 
> Camera Controls
> 
>                   auto_exposure 0x009a0901 (menu)   : min=0 max=1 default=1 value=1 (Manual Mode)
> 
> Image Source Controls
> 
>               vertical_blanking 0x009e0901 (int)    : min=24 max=30823 step=1 default=24 value=24
>             horizontal_blanking 0x009e0902 (int)    : min=252 max=252 step=1 default=252 value=252 flags=read-only
>                   analogue_gain 0x009e0903 (int)    : min=16 max=1023 step=1 default=32 value=248
> 
> Image Processing Controls
> 
>                  link_frequency 0x009f0901 (intmenu): min=0 max=3 default=0 value=0 (175000000 0xa6e49c0)
>                      pixel_rate 0x009f0902 (int64)  : min=87500000 max=87500000 step=1 default=87500000 value=87500000 flags=read-only

So things appear to be mostly working in that the system recognizes the camera on the i2c bus and the user controls, but streaming video does not appear to be working properly.

Al3XKOoL, 2024-04-23:
Hi.

Maybe related with https://github.com/Joshua-Riek/ubuntu-rockchip/issues/730
Camera stream not work on frigate docker with the new kernel. But i'm using rtsp camera.

simon511000, 2024-04-23:
>  Camera stream not work on frigate docker with the new kernel. But i'm using rtsp camera.

Could you try using the `marca711/frigate:latest-rk` image instead?
That solved the issue for me.

Al3XKOoL, 2024-04-23:
> > Camera stream not work on frigate docker with the new kernel. But i'm using rtsp camera.
> 
> Could you try using the `marca711/frigate:latest-rk` image instead? That solved the issue for me.

Great, i will try latter.

Joshua-Riek, 2024-06-24:
Closing see https://github.com/Joshua-Riek/ubuntu-rockchip/pull/690.

Please open again if this problem still happens.

closed 2024-06-24 by Joshua-Riek