Joshua-Riek/ubuntu-rockchip #742: Orange Pi Pro 5 NIC

drsrinathsridhar, 2024-04-25:
@Joshua-Riek Thank you for your service creating these clean Ubuntu builds for Rockchip devices. I see that you have started adding support for the newly released Orange Pi 5 Pro since 6c1ff10.

We are currently using the Orange Pi 5 and 5B for projects that need PTP hardware timestamping. Do we know if the Orange Pi 5 Pro supports PTP hardware timestamping like the 5 and 5B? I tried the Ubuntu 22.04 xfce images from the manufacturer and don't see hardware timestamping using `sudo ethtool -T <NIC>` on the 5 Pro.

Thank you!

Joshua-Riek, 2024-04-25:
Uhhh the opi5 pro has some large software issues. I'm considering dropping the board as I'm extremely burnt out right now and I just want to release ubuntu 24.04.

Primarily the ethernet driver causes a kernel panic and I can't bring myself to fix  it.

Joshua-Riek, 2024-04-25:
I saw PTP hardware timestamping was not enabled in general, so I enabled it. I will do some testing later today.

https://github.com/Joshua-Riek/linux-rockchip/commit/c2daa578c7348ac1e41b15387caf45972d993df9

drsrinathsridhar, 2024-04-25:
Thank you! I don't know enough about the project to contribute, but we are happy to help test or check changes with the Opi 5 pros.

FWIW, I tried the official Ubuntu and Arch Linux versions from the manufacturer. For the Orange Pi 5, it shows hardware timestamping, but does not show it for OPi 5 Pro. It also shows a different ethernet device (Broadcom for OPi 5, something else for Opi 5 Pro). I guess this doesn't necessarily mean there's no hardware timestamping -- the drivers may not have support.

Joshua-Riek, 2024-04-25:
It may need to be enabled in the ethernet driver yt6801. I'm looking shortly.

Joshua-Riek, 2024-04-25:
From what I can see, the driver probes abilities of the ethernet chip, below is where the driver will get the value for time stamping support. https://github.com/Joshua-Riek/linux-rockchip/blob/c2daa578c7348ac1e41b15387caf45972d993df9/drivers/net/ethernet/motorcomm/fuxi-gmac-common.c#L648-L650
 
I can't find much information about the yt6801 ethernet chip, so i can't determine if the hardware even supports PTP timestamping. This will need some research.

drsrinathsridhar, 2024-04-25:
I have requested one of my students to check directly with Motorcomm if this chip has the timestamping feature.

It seems like the driver you linked is from this [Motorcomm link](https://www.motor-comm.com/product/ethernet-control-chip). On the website (after translation), it says this driver is specifically for the yt6801. If there's some logic in the code, there could be a chance it supports hardware TS.

drsrinathsridhar, 2024-04-29:
Over the weekend, I had some time to patch Orange Pi's kernel to use the latest Motorcomm YT6801 driver and enabled `CONFIG_NETWORK_PHY_TIMESTAMPING=y` in the config. This patched kernel also showed no hardware timestamping.

Could things get worse? Yes, because when I was doing these tests, I realized this chip doesn't even support software timestamping (`software-transmit`). So basically, PTP is out of question on the YT6801 and consequently on the Orange Pi 5 Pro.

We also got confirmation this morning from Motorcomm (Frank Sae) that this chip cannot do any form of PTP.

The Orange Pi 5 Pro was promising for us, but in the end didn't work out. I still think it's a nice board for other projects that don't need PTP.

It's safe to close this issue at this point. Nonetheless, thanks for your efforts, Joshua.

Joshua-Riek, 2024-04-29:
Got it, thank you for reaching out to Motorcomm, this is valuable information that I'm sure other individuals will ask the same question.