Joshua-Riek/ubuntu-rockchip #743: libGL error, failed to load driver: rockchip causing pyQT application using RVIZ to lag

programmeddeath1, 2024-04-25:
Hiii
Thank you so much for this awesome work, it has helped way more than any support the opi software provides!

I am trying to run a pyqt application which uses [RVIZ](https://github.com/ros-visualization/rviz) which is a ROS(Robotic Operating System) 3D visualizer.

When starting the app which loads the 3D model on start it gives the following warnings - 
```
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip
[ INFO] [1714028941.896605237]: Stereo is NOT SUPPORTED
[ INFO] [1714028941.896694679]: OpenGL device: llvmpipe (LLVM 12.0.0, 128 bits)
[ INFO] [1714028941.896733974]: OpenGl version: 3.1 (GLSL 1.4).
libGL error: failed to create dri screen
libGL error: failed to load driver: rockchip 
```
The app does load properly, the pyqt application itself does run smoothly, but when i run certain motions on the simulator, it seems to be lagging a lot. I ran `LIBGL_DEBUG=verbose python3 main.py` and it gives the following detailed error - [file](https://github.com/programmeddeath1/webhost/blob/master/opi_rockhip_driver_error)

What do i need to enable the gpu driver usage or improve the rendering, any help would be appreciated.

Thanks in advance!

Joshua-Riek, 2024-04-25:
The program is using llvmpipe (no gpu acceleration). Is this program being run from docker or from the desktop environment?

programmeddeath1, 2024-04-25:
Yes This one currently is running from a docker. I have setup my environment in (ubuntu20.04) docker for the latest ubuntu 22.04 os  from the v1.33 release. ROS 1 currently only supports ubuntu 20.04.

I have setup a similar environment in the official ubuntu 20.04 version 1.18 release by them in Feb. It also gives the same error with rockchip driver during the RVIZ load.(I havent seen the verbose error i can check and add in sometime) But this i suppose is because they havent added hardware acceleration altogether in their official versions.

Joshua-Riek, 2024-04-25:
That is the problem, running the software in docker.

The system requires multiple user space hacks to support hardware acceleration, you would need to modify and hack the docker image. This won't be an easy task, especially when trying to use Ubuntu 20.04. 

I do not know much about ROS, but I would try to run your software outside of docker.

programmeddeath1, 2024-04-25:
I had used an ubuntu 20 image of yours like a year or so back. but they do not seem to be available on your github anymore. Would this work on any of your older ubuntu 20.04 releases you had before?

If that option isn't possible, I do need to get this to work in my docker if possible, I see that its trying to search for those directories for the driver libraries, would loading the host directories of /etc/ and other needed directories be enough or are there any deeper inter linkings that I might need to do?

If you have a guide or reference of the steps needed to get this to work I'm willing to dig deep and try to get it done

programmeddeath1, 2024-04-26:
I tried it on the older orangepi ubuntu 20.04 os v1.6 . It gives the same error of unable to find gpu drivers. 

https://github.com/programmeddeath1/webhost/blob/master/libgl_error_rviz.txt

I feel like the RVIZ application is looking for gpu drivers to be available in a specific location, and maybe giving it the path to the default driver locations would help or will installing panfrost mali gpu drivers in the ubuntu from opi help fix this?

ajay01994, 2024-04-28:
@Joshua-Riek could you help us in anyway ?

Joshua-Riek, 2024-04-28:
I'm sorry but I have no interest in this, it's too much time and effort on my part.

ajay01994, 2024-05-05:
Hmm, interesting! Indeed, it seems the issue lies with Docker's inability to access the GPU in RK3588. Creating something akin to the NVIDIA Container Toolkit (https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html) for future Docker support could be a promising solution. Thanks for bringing this up!

avi951, 2024-09-12:
Yes, Its takes too much time to debug, still facing same issue in my case rk3588 rockchip.
@ajay01994 Have you got any idea to solve this issue ?

RomenKlaassen, 2024-11-29:
Hi all,
I will first say that I was facing the exact same problem not too long ago, but got it fixed by creating my own Ubuntu 22.04 Docker base-image from the base-image provided by Joshua-Riek. This will force the whole docker container environment to use the "Libmali Panfrost" driver.

It took me WAY TOO LONG to debug this, but I noticed that outside of my docker container (on the linux desktop environment), I ran the command:

glxinfo | grep "OpenGL"

and it showed me the Panfrost libmali iGPU driver. This told me that it correctly uses the iGPU for rendering.
I then tried to get it working inside my Docker container environment in every single way possible, but failed horribly.

It suddenly hit me that I could try and recreate my whole Ubuntu 22.04 Linux Base-image and convert it to an image to use as a Docker Base-image. This seemed to have fixed the problem....

Good luck!

programmeddeath1, 2024-11-30:
Hey that seems to be a good solution! 

We worked around this issue by just building ros 1 in ubuntu 22 itself from source. So that bypassed the need for running it through docker.

closed 2024-11-30 by programmeddeath1

RomenKlaassen, 2024-11-30:
I did a little more research and found the following website which sets up hardware acceleration in a docker container (originally to use jellyfin, but the setup should be the same for any hardware accelerated container, I think):

https://akashrajpurohit.com/blog/setup-jellyfin-with-hardware-acceleration-on-orange-pi-5-rockchip-rk3558/