Joshua-Riek/ubuntu-rockchip #748: Default user and password for 22.04-beta

enmanuelmoreira, 2024-04-26:
What is the default user and password for 22.04-beta?

Joshua-Riek, 2024-04-26:
there is no 22.04 beta.. but try ubuntu and ubuntu..

closed 2024-04-26 by Joshua-Riek

enmanuelmoreira, 2024-04-26:
I am sorry, It's 24.04 beta, I tried with ubuntu:ubuntu but it doesn't work

enmanuelmoreira, 2024-04-26:
Hi I sorry bother you, I was trying to install ubuntu24.04 beta on my orange pi plus 5, but you closed and marked by spam my conversation. Thanks for you help, I really appreciate that

nilo85, 2024-04-28:
As i understand it, there is / was some issue with cloudinit and timing depending on what build you use. And that the cloud init can be slow / issues. 

Looking at commit logs it looks like we are getting closer and closer to a release candidate / version of 24.04. So I would recommend 
* Try building your one image if you want the latest 24.04 stuff super soon based on latest master 
* Wait a few days and see if there is a release of 24.04 
* Try 22.04 meanwhile.