Joshua-Riek/ubuntu-rockchip #753: Ubuntu 24.04 Beta crashes after reboot

cantalupo555, 2024-04-29:
When I'm using this Beta image on the memory card (/dev/mmcblk0) I haven't had any problems, I can restart several times.

The problem happens when I'm using NVMe (/dev/nvme0n1).
It installs normally, but if I restart the machine Ubuntu 24.04 Beta simply does not start.

Has anyone else gone through this?

Joshua-Riek, 2024-04-29:
Did you update the SPI? Almost 99% of crashing / boot issues is because of a bad power supply or the SPI is not updated.

cantalupo555, 2024-04-29:
Thanks for the feedback.  In reality, my problem is not the power supply, how can I update the SPI?  Would it be /dev/mtdblock0?

Joshua-Riek, 2024-04-29:
Please review the below WiKi page for Ubuntu 24.04:

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#install-u-boot-to-the-spi-flash

Joshua-Riek, 2024-04-30:
Awesome, enjoy!

closed 2024-04-30 by Joshua-Riek