Joshua-Riek/ubuntu-rockchip #754: 24.04 not using overlays from this fork

dev-board-tech, 2024-04-29:
At build time is using a prebuild package to install device overlays instead of the ones in this fork even if the log shows that it installs the ones from this fork.

Joshua-Riek, 2024-04-29:
I'm a little confused on what the issue is. The device tree overlays are included in the firmware dir of the system.

dev-board-tech, 2024-04-29:
The issue is that build-kernet.sh script clone the linux-rockip repository that include the device tree dtb and dtbo's placed in "build/linux-rockchip/arch/arm64/boot/dts/rockchip" when the image is build the content of that directory need to be copied to "/lib/firmware/6.1.0-1012-rockchip/device-tree/rockchip" in ubuntu 24 case.

The changes made in "linux-rockchip/arch/arm64/boot/dts/rockchip" directory are not included in final image, instead in final image are included dtb's and dtbo's from a downloaded package, I think from this package: "linux-rockchip-headers-6.1.0-1012".

Even if I manually copy the new device tree to "build/ubuntu-live-build/build/chroot/lib/firmware/6.1.0-1012-rockchip/device-tree/rockchip" to pick them and build the image is still install the device tree from "linux-rockchip-headers-6.1.0-1012"

The log show that the device tree is correctly build and installed but the changes don't make it in the image.

Joshua-Riek, 2024-04-29:
The build-kernel.sh script no longer is used for Ubuntu 24.04 and will be removed when I get the time. I now follow the upstream kernel packaging from Ubuntu. To be honest the kernel packaging is weird and complex, users will need to do some digging to properly modify they're on kernel. 

You can checkout the kernel source code and build as below:
```
export CC=aarch64-linux-gnu-gcc
export $(dpkg-architecture -aarm64)
export CROSS_COMPILE=aarch64-linux-gnu-

LANG=C fakeroot debian/rules clean binary-rockchip
```

dev-board-tech, 2024-04-29:
For my needs I can copy the newly compiled "rk3588s-9tripod-linux.dtb" "rk3588s-indiedroid-nova.dtb" ( cuz the device tree is correctly compiled ) and my other dtbo's to firmware after the image has been generated, but is good to know that any changes in the device tree are useless cuz are not included into the image.

Joshua-Riek, 2024-04-29:
Hey, don't blame me, I'm following upstream to provide a better product to users. If there is a problem with upstream methods, where something is redundant, I would recommend trying to change it upstream.

dev-board-tech, 2024-04-29:
I will never blame you, instead I thank you very much for your great work, your build is the closest to the desktop one, the rest are cut down or stripped of functionalities, out there I see builds that I don't know why...

vrobinyh, 2024-04-30:
Hi all,
I try boot from sata but dosn't work :(
I add this line  for the extlinux.conf 
fdtoverlays  /lib/firmware/6.1.0-1012-rockchip/device-tree/rockchip/overlay/orangepi-5-sata.dtbo
dosn't work the boot.
(22.04 work perfectly the boot from sata)
Please help
thank you

peytonicmaster6, 2024-05-01:
@vrobinyh You'll want to follow the device tree overlay guide for 24.04 in the [wiki](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#using-a-device-tree-overlay).

closed 2024-12-16 by github-actions[bot]