Joshua-Riek/ubuntu-rockchip #755: 24.04 initramfs tries to install to wrong device.?

nilo85, 2024-04-29:
Running a fresh build from yesterday ("remove beta" commit)

Just ran apt update followed by apt upgrade and got stuck here

```
Unpacking rockchip-firmware (20231105.git2107be54-17~noble) over (20231105.git2107be54-15~noble) ...
Setting up rockchip-firmware (20231105.git2107be54-17~noble) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-1012-rockchip
I: The initramfs will attempt to resume from /dev/nvme0n1p1
I: (UUID=97c80205-5cdd-4204-b65a-096a1828f5ec)
I: Set the RESUME variable to override this.
```

Now, the problem is my system is installed on emmc, /dev/nvme0n1p1 would be my swap partition.

Not sure what to do here =)

nilo85, 2024-04-29:
Full output:
```
sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  curl less libc-bin libc-dev-bin libc-devtools libc6 libc6-dev libcurl3t64-gnutls libcurl4t64 libgnutls30t64 locales rockchip-firmware
12 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
11 standard LTS security updates
Need to get 18.4 MB of archives.
After this operation, 97.3 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libc-devtools arm64 2.39-0ubuntu8.1 [27.8 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libc6-dev arm64 2.39-0ubuntu8.1 [1596 kB]
Get:3 http://ppa.launchpad.net/jjriek/rockchip/ubuntu noble/main arm64 rockchip-firmware all 20231105.git2107be54-17~noble [7151 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libc-dev-bin arm64 2.39-0ubuntu8.1 [19.7 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libc6 arm64 2.39-0ubuntu8.1 [2774 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libc-bin arm64 2.39-0ubuntu8.1 [591 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libgnutls30t64 arm64 3.8.3-1.1ubuntu3.1 [935 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 less arm64 590-2ubuntu2.1 [142 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 locales all 2.39-0ubuntu8.1 [4234 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 curl arm64 8.5.0-2ubuntu10.1 [222 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libcurl4t64 arm64 8.5.0-2ubuntu10.1 [333 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 libcurl3t64-gnutls arm64 8.5.0-2ubuntu10.1 [327 kB]
Fetched 18.4 MB in 3s (5888 kB/s)
Preconfiguring packages ...
(Reading database ... 73743 files and directories currently installed.)
Preparing to unpack .../libc-devtools_2.39-0ubuntu8.1_arm64.deb ...
Unpacking libc-devtools (2.39-0ubuntu8.1) over (2.39-0ubuntu8) ...
Preparing to unpack .../libc6-dev_2.39-0ubuntu8.1_arm64.deb ...
Unpacking libc6-dev:arm64 (2.39-0ubuntu8.1) over (2.39-0ubuntu8) ...
Preparing to unpack .../libc-dev-bin_2.39-0ubuntu8.1_arm64.deb ...
Unpacking libc-dev-bin (2.39-0ubuntu8.1) over (2.39-0ubuntu8) ...
Preparing to unpack .../libc6_2.39-0ubuntu8.1_arm64.deb ...
Unpacking libc6:arm64 (2.39-0ubuntu8.1) over (2.39-0ubuntu8) ...
Setting up libc6:arm64 (2.39-0ubuntu8.1) ...
(Reading database ... 73743 files and directories currently installed.)
Preparing to unpack .../libc-bin_2.39-0ubuntu8.1_arm64.deb ...
Unpacking libc-bin (2.39-0ubuntu8.1) over (2.39-0ubuntu8) ...
Setting up libc-bin (2.39-0ubuntu8.1) ...
(Reading database ... 73743 files and directories currently installed.)
Preparing to unpack .../libgnutls30t64_3.8.3-1.1ubuntu3.1_arm64.deb ...
Unpacking libgnutls30t64:arm64 (3.8.3-1.1ubuntu3.1) over (3.8.3-1.1ubuntu3) ...
Setting up libgnutls30t64:arm64 (3.8.3-1.1ubuntu3.1) ...
(Reading database ... 73743 files and directories currently installed.)
Preparing to unpack .../0-less_590-2ubuntu2.1_arm64.deb ...
Unpacking less (590-2ubuntu2.1) over (590-2ubuntu2) ...
Preparing to unpack .../1-locales_2.39-0ubuntu8.1_all.deb ...
Unpacking locales (2.39-0ubuntu8.1) over (2.39-0ubuntu8) ...
Preparing to unpack .../2-curl_8.5.0-2ubuntu10.1_arm64.deb ...
Unpacking curl (8.5.0-2ubuntu10.1) over (8.5.0-2ubuntu10) ...
Preparing to unpack .../3-libcurl4t64_8.5.0-2ubuntu10.1_arm64.deb ...
Unpacking libcurl4t64:arm64 (8.5.0-2ubuntu10.1) over (8.5.0-2ubuntu10) ...
Preparing to unpack .../4-libcurl3t64-gnutls_8.5.0-2ubuntu10.1_arm64.deb ...
Unpacking libcurl3t64-gnutls:arm64 (8.5.0-2ubuntu10.1) over (8.5.0-2ubuntu10) ...
Preparing to unpack .../5-rockchip-firmware_20231105.git2107be54-17~noble_all.deb ...
Unpacking rockchip-firmware (20231105.git2107be54-17~noble) over (20231105.git2107be54-15~noble) ...
Setting up rockchip-firmware (20231105.git2107be54-17~noble) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-1012-rockchip
I: The initramfs will attempt to resume from /dev/nvme0n1p1
I: (UUID=4abb0abf-78e4-4ad1-a317-4e3882631fd6)
I: Set the RESUME variable to override this.
Setting up libcurl4t64:arm64 (8.5.0-2ubuntu10.1) ...
Setting up libcurl3t64-gnutls:arm64 (8.5.0-2ubuntu10.1) ...
Setting up less (590-2ubuntu2.1) ...
Setting up locales (2.39-0ubuntu8.1) ...
Generating locales (this might take a while)...
  en_US.UTF-8... done
Generation complete.
Setting up curl (8.5.0-2ubuntu10.1) ...
Setting up libc-dev-bin (2.39-0ubuntu8.1) ...
Setting up libc-devtools (2.39-0ubuntu8.1) ...
Setting up libc6-dev:arm64 (2.39-0ubuntu8.1) ...
Processing triggers for man-db (2.12.0-4build2) ...
Processing triggers for libc-bin (2.39-0ubuntu8.1) ...
Scanning processes...
Scanning candidates...
Scanning processor microcode...
Scanning linux images...

Running kernel seems to be up-to-date.

The processor microcode seems to be up-to-date.

Restarting services...
 systemctl restart cron.service multipathd.service polkit.service rsyslog.service snapd.service ssh.service systemd-journald.service systemd-networkd.service systemd-resolved.service systemd-timesyncd.service systemd-udevd.service udisks2.service

Service restarts being deferred:
 systemctl restart ModemManager.service
 /etc/needrestart/restart.d/dbus.service
 systemctl restart getty@tty1.service
 systemctl restart serial-getty@ttyFIQ0.service
 systemctl restart systemd-logind.service
 systemctl restart unattended-upgrades.service

No containers need to be restarted.

User sessions running outdated binaries:
 ubuntu @ session #1: apt[3678], sshd[1046,1154]
 ubuntu @ user manager service: systemd[1050]

No VM guests are running outdated hypervisor (qemu) binaries on this host.
```

nilo85, 2024-04-29:
Interesting it did now come up again after quite some time, not sure how healthy it is =)

Joshua-Riek, 2024-04-29:
It may happen when you install another OS alongside the first one. Means that swap partition has changed UUID and you need to adjust that in fstab.

What is the output of /etc/fstab on your system?

Joshua-Riek, 2024-04-29:
I think I know a fix, but this is a damn good find.

nilo85, 2024-04-29:
```
cat /etc/fstab
# <file system>     <mount point>  <type>  <options>   <dump>  <fsck>
UUID=78c158f4-88a6-43d0-88c5-33c5aeba24f6 /              ext4    defaults,x-systemd.growfs    0       1
UUID=97c80205-5cdd-4204-b65a-096a1828f5ec swap	swap	defaults	0	0
UUID=4cff537f-4179-4d96-bfab-26dd59d0f2bb /media/data	ext4	defaults	0	2
```

Joshua-Riek, 2024-04-29:
And if you check `sudo blkid`, that gives you actual UUIDs of your swaps, swapon -s will show which one is active, then compare its UUID to the one in /etc/fstab. Does the UUID match?

nilo85, 2024-04-29:
Sorry wrong machine, the fstab was from the one that was rebooting and offline when I wrote this issue.

fstab on machine on above dump
```
ubuntu@k3s-2:~$ cat /etc/fstab
# <file system>     <mount point>  <type>  <options>   <dump>  <fsck>
UUID=78c158f4-88a6-43d0-88c5-33c5aeba24f6 /              ext4    defaults,x-systemd.growfs    0       1
UUID=4abb0abf-78e4-4ad1-a317-4e3882631fd6 swap	swap	defaults	0	0
UUID=d80eff53-a032-4b59-a79b-b768bfe9237f /media/data	ext4	defaults	0	2
```

```
ubuntu@k3s-2:~$ sudo blkid /dev/mmcblk0*
/dev/mmcblk0: PTUUID="f25743e9-5807-4c33-8c28-8a33ab0532a5" PTTYPE="gpt"
/dev/mmcblk0p1: LABEL_FATBOOT="CIDATA" LABEL="CIDATA" UUID="D142-0EAD" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="primary" PARTUUID="e04ca42e-fbaa-4440-9614-b39d71720cf9"
/dev/mmcblk0p2: LABEL="cloudimg-rootfs" UUID="78c158f4-88a6-43d0-88c5-33c5aeba24f6" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="bb9aadb2-a5d7-437d-9e7a-dd5b9cb26659"
ubuntu@k3s-2:~$ sudo blkid /dev/nvme0n1*
/dev/nvme0n1: PTUUID="4752f538-4eff-7440-a94c-fde856163f80" PTTYPE="gpt"
/dev/nvme0n1p1: UUID="4abb0abf-78e4-4ad1-a317-4e3882631fd6" TYPE="swap" PARTUUID="98ecc318-2ea1-4ad8-b31f-98bb176f7e09"
/dev/nvme0n1p2: UUID="d80eff53-a032-4b59-a79b-b768bfe9237f" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="785a7d79-a436-4d73-a042-d26b0144fb41"
```

nilo85, 2024-04-29:
As far as I can see the UUIDs looks right in fstab. and they are all unique within per machine (rootfs same on both machines as it was precalculated )

nilo85, 2024-04-29:
@Joshua-Riek Yeah, my idea was to have swap + data on SSD, and OS on emmc for hopefully easier upgrades in the future =)

Added the RESUME=none now and trying "sudo update-initramfs -u -k all" (at this stage its a toy anyway)

Joshua-Riek, 2024-04-29:
I see, so looks like adding RESUME=none fixed the problem?

nilo85, 2024-04-29:
Looks like =) It kind of scared me pointing to the wrong disk but seems it might not have done any harm, just needed longer than normal to boot afterwards

Joshua-Riek, 2024-04-29:
Gotcha, I don't think this is something i can permanently fix it would require a user to modify their system since this is a specific use case.

nilo85, 2024-04-29:
So I guess maybe its just me having a weird setup, and as you pointed out, another post with similar issue so guess its a general ubuntu thing and not a ubuntu-rockchip thing.

Guess this can be closed as invalid then, super thanks for all your work and support =)

Joshua-Riek, 2024-04-29:
The official release should be very soon, at the latest tmr. I'm waiting on chrome to finish building.

nilo85, 2024-04-29:
I closed it, hope I didnt mess too much with your notifications and it ended up having a desirable close status =) (only had 2 options)

closed 2024-04-29 by nilo85

Joshua-Riek, 2024-04-29:
Haha no worries!