Joshua-Riek/ubuntu-rockchip #758: Orange Pi 5 Plus HMDI In

IainG, 2024-05-01:
A recent update seems to have broken the hdmi in capability at a driver level - there is no /dev/video device, so therefore `v4l2-ctl --list-devices` fails with `Cannot open device /dev/video0, exiting. `

There are quite a few hdmi related errors in dmesg, perhaps some of these are relevant (however I'm also having a few other screen related issues, :

`rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy pll locked!`
`phy phy-fed70000.hdmiphy.10: phy poweron failed --> -22`
`dwhdmi-rockchip fdea0000.hdmi: dw_hdmi_qp_setup hdmi set operation mode failed`

or perhaps the only message related to hdmirx:

```
[    7.504136] rockchip-dmc dmc: hdmirx_rate = 2112000000
[    7.504145] rockchip-dmc dmc: failed to get vop bandwidth to dmc rate
```

The test bash script below used to work fine, unfortunately I'm not aware which update seems to have broken the functionality.

```
!/bin/bash
trap 'onCtrlC' INT
function onCtrlC () {
echo 'Ctrl+C is captured'
killall gst-launch-1.0
exit 0
}

device_id=$(v4l2-ctl --list-devices | grep -A1 hdmirx | grep -v hdmirx | awk -F ' ' '{print $NF}')
v4l2-ctl -d $device_id --set-dv-bt-timings query 2>&1 > /dev/null
width=$(v4l2-ctl -d $device_id --get-dv-timings | grep "Active width" |awk -F ' ' '{print $NF}')
heigh=$(v4l2-ctl -d $device_id --get-dv-timings | grep "Active heigh" |awk -F ' ' '{print $NF}')

es8388_card=$(aplay -l | grep "es8388" | cut -d ':' -f 1 | cut -d ' ' -f 2)
hdmi0_card=$(aplay -l | grep "hdmi0" | cut -d ':' -f 1 | cut -d ' ' -f 2)
hdmi1_card=$(aplay -l | grep "hdmi1" | cut -d ':' -f 1 | cut -d ' ' -f 2)
hdmiin_card=$(arecord -l | grep "hdmiin" | cut -d ":" -f 1 | cut -d ' ' -f 2)

if [[ $XDG_SESSION_TYPE == wayland ]]; then
DISPLAY=:0.0 gst-launch-1.0 v4l2src device=${device_id} ! videoconvert \
! videoscale ! video/x-raw,width=1280,height=720 \
! waylandsink sync=false 2>&1 > /dev/null &
else
DISPLAY=:0.0 gst-launch-1.0 v4l2src device=${device_id} io-mode=4 ! videoconvert \
! video/x-raw,format=NV12,width=${width},height=${heigh} \
! videoscale ! video/x-raw,width=1280,height=720 \
! autovideosink sync=false 2>&1 > /dev/null &

fi

gst-launch-1.0 alsasrc device=hw:${hdmiin_card},0 ! audioconvert ! audioresample ! queue \
! tee name=t ! queue ! alsasink device="hw:${hdmi0_card},0" \
t. ! queue ! alsasink device="hw:${hdmi1_card},0" \
t. ! queue ! alsasink device="hw:${es8388_card},0" &

while true
do
sleep 10
done
```

Any advice on what to roll back to, or what could have disabled the hdmi would be appreciated!

nilo85, 2024-05-01:
There is an issue with the HDMI in that caused the run with >1 in load with heat and performance issues as a result. (https://github.com/Joshua-Riek/ubuntu-rockchip/issues/606)
So short term this is disabled by default.

@Joshua-Riek added some sort of overlay to enable it, dont know how that works though but suspect you need to enable it to get it working again
https://github.com/Joshua-Riek/linux-rockchip/commit/03b1f58bd995f0d8a6701d868d3fbfa06ffed5c7

nilo85, 2024-05-01:
I guess you need to do this kind of thing:
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

```
overlays=rk3588-hdmirx
```

IainG, 2024-05-01:
> I guess you need to do this kind of thing: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays
> 
> ```
> overlays=rk3588-hdmirx
> ```

Thanks so much @nilo85, that does indeed reenable the hdmi in device - I guess I'll need to keep an eye on load and see how it goes!

closed 2024-05-01 by IainG

feiyax, 2024-09-04:
Thanks to this thread, i also got HDMI Rx to show up.

One thing differed from the last comment is that the device tree overlay usage has changed. Here's the latest description for ubuntu 24.04: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#using-a-device-tree-overlay 

I ended first locating the device tree overlay: 

```
root@yax-opi5p:~# find /lib/firmware/ -name '*rk3588-hdmirx*'
/lib/firmware/6.1.0-1021-rockchip/device-tree/rockchip/overlay/rk3588-hdmirx.dtbo
```

Then specify that in file `/etc/default/u-boot` in this way: 

```
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-hdmirx.dtbo"
```

Then `sudo u-boot-update` and reboot.

I did it with:

```
root@yax-opi5p:~# uname -a
Linux yax-opi5p 6.1.0-1021-rockchip #21-Ubuntu SMP Mon Jul 29 03:52:32 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
root@yax-opi5p:~# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:        24.04
Codename:       noble
root@yax-opi5p:~# cat /proc/version
Linux version 6.1.0-1021-rockchip (buildd@bos03-arm64-074) (aarch64-linux-gnu-gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #21-Ubuntu SMP Mon Jul 29 03:52:32 UTC 2024
```