Joshua-Riek/ubuntu-rockchip #759: 24.04: OrangePi5Plus dma failed

dongxuanlb, 2024-05-01:
my board is Oranage Pi 5 Plus 32G + 256GB emmc with the latest orangepi5p 6.1.0-1012-rockchip 24.04 version img.

dma seems init failed.

```
dx@orangepi5p:~$ ll /dev/dma_heap/
total 0
drwxr-xr-x  2 root root      80 Jan  1  1970 ./
drwxr-xr-x 19 root root    3920 May  1 14:17 ../
crw-rw----  1 root video 251, 1 May  1 15:02 cma
crw-rw----  1 root video 251, 0 May  1 15:02 system
```

```
dx@orangepi5p:~$ dmesg|grep rga
[    7.125195] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga30-supply from device tree
[    7.125221] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga30-supply property in node /power-management@fd8d8000/power-controller failed
[    7.125653] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga31-supply from device tree
[    7.125675] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up rga31-supply property in node /power-management@fd8d8000/power-controller failed
[    7.985400] rga3_core0 fdb60000.rga: Adding to iommu group 2
[    7.986178] rga: rga3_core0, irq = 49, match scheduler
[    7.986886] rga: rga3_core0 hardware loaded successfully, hw_version:3.0.76831.
[    7.987509] rga: rga3_core0 probe successfully
[    7.988309] rga3_core1 fdb70000.rga: Adding to iommu group 3
[    7.988959] rga: rga3_core1, irq = 50, match scheduler
[    7.989659] rga: rga3_core1 hardware loaded successfully, hw_version:3.0.76831.
[    7.990254] rga: rga3_core1 probe successfully
[    7.991023] rga: rga2, irq = 166, match scheduler
[    7.991713] rga: rga2 hardware loaded successfully, hw_version:3.2.63318.
[    7.992313] rga: rga2 probe successfully
[    7.992965] rga_iommu: IOMMU binding successfully, default mapping core[0x1]
[    7.993682] rga: Module initialized. v1.3.1
[ 8804.811437] rga_dma_buf: Fail to get dma_buf from fd[779303030], ret[-9]
[ 8804.811473] rga_mm: rga_mm_map_dma_buffer core[1] map dma buffer error!
[ 8804.811485] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 8804.811495] rga: buffer[0] mm import buffer failed! memory = 0x2e733876, type = dma_fd(0x0)
[ 8804.811931] rga_dma_buf: Fail to get dma_buf from fd[577135976], ret[-9]
[ 8804.811953] rga_mm: rga_mm_map_dma_buffer core[1] map dma buffer error!
[ 8804.811964] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 8804.811974] rga: buffer[0] mm import buffer failed! memory = 0x22666568, type = dma_fd(0x0)
[ 8804.812297] rga_dma_buf: Fail to get dma_buf from fd[1634208266], ret[-9]
[ 8804.812317] rga_mm: rga_mm_map_dma_buffer core[1] map dma buffer error!
[ 8804.812328] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 8804.812337] rga: buffer[0] mm import buffer failed! memory = 0x61680a0a, type = dma_fd(0x0)

```

```
dx@orangepi5p:~$ dmesg|grep dma
[    4.540439] OF: reserved mem: initialized node cma, compatible id shared-dma-pool
[    7.117120] dma-pl330 fea10000.dma-controller: Loaded driver for PL330 DMAC-241330
[    7.117141] dma-pl330 fea10000.dma-controller: 	DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
[    7.118177] dma-pl330 fea30000.dma-controller: Loaded driver for PL330 DMAC-241330
[    7.118193] dma-pl330 fea30000.dma-controller: 	DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
[    7.119243] dma-pl330 fed10000.dma-controller: Loaded driver for PL330 DMAC-241330
[    7.119260] dma-pl330 fed10000.dma-controller: 	DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
[   10.348481] hailo 0000:01:00.0: Probing: Enabled 64 bit dma
[   10.348489] hailo 0000:01:00.0: Probing: Using userspace allocated vdma buffers
[ 8804.811437] rga_dma_buf: Fail to get dma_buf from fd[779303030], ret[-9]
[ 8804.811473] rga_mm: rga_mm_map_dma_buffer core[1] map dma buffer error!
[ 8804.811485] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 8804.811495] rga: buffer[0] mm import buffer failed! memory = 0x2e733876, type = dma_fd(0x0)
[ 8804.811931] rga_dma_buf: Fail to get dma_buf from fd[577135976], ret[-9]
[ 8804.811953] rga_mm: rga_mm_map_dma_buffer core[1] map dma buffer error!
[ 8804.811964] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 8804.811974] rga: buffer[0] mm import buffer failed! memory = 0x22666568, type = dma_fd(0x0)
[ 8804.812297] rga_dma_buf: Fail to get dma_buf from fd[1634208266], ret[-9]
[ 8804.812317] rga_mm: rga_mm_map_dma_buffer core[1] map dma buffer error!
[ 8804.812328] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 8804.812337] rga: buffer[0] mm import buffer failed! memory = 0x61680a0a, type = dma_fd(0x0)
```

nilo85, 2024-05-01:
I guess this is not really noticable on server build, but I get the same boot output, the dev/dma_heap/system-uncached-dma32 and /dev/dma_heap/system-dma32 does not exist on my machine ether.. (however no clue about this =D) just wanted to confirm it seems to be >N1 issue

nyanmisaka, 2024-05-01:
https://github.com/JeffyCN/mirrors/issues/18#issuecomment-1977774286

They are removed intentionally in rockchip linux 6.1 kernel. Use `/dev/dma_heap/system` or drm gem instead.

Joshua-Riek, 2024-05-01:
Yeah

closed 2024-05-01 by Joshua-Riek