Joshua-Riek/ubuntu-rockchip #765: Missing libtglapi-mesa break upgrades

terala, 2024-05-02:
I am running into this issue when I upgrade packages:
```
╰─$ sudo apt upgrade -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following upgrades have been deferred due to phasing:
  distro-info-data
The following packages will be upgraded:
  libegl-mesa0 libgbm1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 mesa-opencl-icd mesa-va-drivers
  mesa-vulkan-drivers
8 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Need to get 8713 kB of archives.
After this operation, 0 B of additional disk space will be used.
Err:1 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 libegl-mesa0 arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:2 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 libgbm1 arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:3 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 libgl1-mesa-dri arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:4 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 libglx-mesa0 arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:5 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 libglapi-mesa arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:6 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 mesa-opencl-icd arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:7 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 mesa-va-drivers arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
Err:8 http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble/main arm64 mesa-vulkan-drivers arm64 1:23.0.5-0ubuntu1~panfork~git221210.120202c6757~noble4
  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/libegl-mesa0_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/libgbm1_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/libgl1-mesa-dri_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/libglx-mesa0_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/libglapi-mesa_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/mesa-opencl-icd_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/mesa-va-drivers_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Failed to fetch http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu/pool/main/m/mesa/mesa-vulkan-drivers_23.0.5-0ubuntu1%7epanfork%7egit221210.120202c6757%7enoble4_arm64.deb  404  Not Found [IP: 185.125.190.80 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Is there a way to fix this?

Joshua-Riek, 2024-05-02:
Looks like an apt server is down. Try again in an hour and it should work fine.

terala, 2024-05-02:
It isn't resolved. This is going on for a couple of days already.

Joshua-Riek, 2024-05-03:
Is this ubuntu 24.04 beta? You may have removed the ppa pin by accident.

terala, 2024-05-03:
This is on 24.04 beta. PPA is still in the list:
```
╭─ravi@opi5plus
╰─$ sudo apt-add-repository --list 
deb http://ppa.launchpad.net/jjriek/rockchip/ubuntu noble main 

deb http://ppa.launchpad.net/jjriek/rockchip-multimedia/ubuntu noble main 

deb http://ppa.launchpad.net/jjriek/panfork-mesa/ubuntu noble main 
```

Joshua-Riek, 2024-05-03:
Hmm id recommend re-installing with the final release. I added some meta packages to assist with fixing problems with updates. 

Additionally, during this time, I fixed an issue with mesa, there was a large problem with dependencies due to the t64 migration.

terala, 2024-05-03:
I wanted to let you know of this issue before I upgrade. I've been using the beta release for the last couple of weeks as my main driver and have been quite happy with it. Please accept my kudos!

closed 2024-05-03 by terala