Joshua-Riek/ubuntu-rockchip #768: Orange Pi 5 Pro Ubuntu 24.04 not booting

KlimRider, 2024-05-03:
The Board OPI 5 Pro not booting with 24.04 image (nor eMMC module nor SD card)
The all i can see is Kerrnel panic
The green and blue LEDs not blink at all or blink couple times
Sometimes the leds just start bright lighting (no blink) and no video/
I think this was i HW issue, but change eversing not to go.
The original image with kernel 5.10.160 booting well

Joshua-Riek, 2024-05-03:
I've noticed this several times, If I recall correctly it's related to PD or the Broadcom wireless driver. Are you able to provide the boot logs that show the kernel panic?

KlimRider, 2024-05-03:
Where i can find that logs?
I mount those SD card and cannot fin any crashlog

Joshua-Riek, 2024-05-03:
You would need to connect via UART to get the logs.

KlimRider, 2024-05-03:
OK, i need few minutes to find my UART adapter

salier, 2024-05-03:
I also found the same problem, I've been tinkering around for almost a week. Whether upgrading from 5.1 to 6.1, installing GPU open source drivers, installing Wayroid, or even installing GNOME, they cannot be used. I ..............I hope it's just my own misfortune. I would like to ask if the 6.1 kernel NPU of the 3588/S is stable and reliable. I hope that the 6.1 mainline adaptation can reduce my difficulties. I feel very uncomfortable wasting almost two hundred hours on this now.If you can teach me how to use UART, I can assist you.

KlimRider, 2024-05-03:
Attached are the logs. 
[eMMC.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15201847/eMMC.log)
[TFC.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15201849/TFC.log)
It's strange that each time the logs end in different places. In the logs, the device was rebooted once using the RESET button. The first part is the actual boot, the second after the RESET button.
eMMC file - booting from.  eMMC module
TFC file - botting from TF card

Joshua-Riek, 2024-05-03:
> Attached are the logs. [eMMC.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15201847/eMMC.log) [TFC.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15201849/TFC.log) It's [...]

Thanks, from the below infomration it looks to be either related to a broken PD implementation or the SPL + DDR blob are bad. Do you have a static 5v 4a power supply?
```
[    7.842676] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[    7.842697] Mem abort info:
[    7.842700]   ESR = 0x0000000096000004
[    7.842703]   EC = 0x25: DABT (current EL), IL = 32 bits
[    7.842708]   SET = 0, FnV = 0
[    7.842710]   EA = 0, S1PTW = 0
[    7.842712]   FSC = 0x04: level 0 translation fault
[    7.842716] Data abort info:
[    7.842718]   ISV = 0, ISS = 0x00000004
[    7.842720]   CM = 0, WnR = 0
[    7.842722] user pgtable: 4k pages, 48-bit VAs, pgdp=000000010eb2d000
[    7.842727] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000
[    7.842742] Internal error: Oops: 0000000096000004 [#1] SMP
[    7.842750] Modules linked in:
[    7.842760] CPU: 3 PID: 261 Comm: (udev-worker) Not tainted 6.1.0-1012-rockchip #12-Ubuntu
[    7.842767] Hardware name: Orange Pi 5 Pro (DT)
[    7.842772] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    7.842780] pc : __cmpxchg_case_acq_64+0x8/0x28
[    7.842807] lr : mutex_lock+0x34/0x54
[    7.842821] sp : ffff80000c3e3cd0
[    7.842824] x29: ffff80000c3e3cd0 x28: 0000000000000000 x27: ffff8000092e3d20
[    7.842833] x26: 0000000000000000 x25: 000000000000000e x24: ffff00010ea7c000
[    7.842842] x23: 0000000000000001 x22: ffff80000c3e3d98 x21: ffff00010ea7c040
[    7.842849] x20: 0000000000000000 x19: 0000000000000000 x18: ffff80000e6a5048
[    7.842857] x17: 0000000000000000 x16: 0000000000000000 x15: 0000ffffeea95fd8
[    7.842865] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[    7.842873] x11: 0000000000000000 x10: 0000000000000000 x9 : ffff800009287218
[    7.842880] x8 : 0000000000000000 x7 : 0000000000000000 x6 : ffff80000c3e3da8
[    7.842889] x5 : ffff80000c3e3da8 x4 : 0000000000000001 x3 : ffff00010ee8c5b8
[    7.842897] x2 : ffff00010ec12000 x1 : 0000000000000000 x0 : 0000000000000000
[    7.842907] Call trace:
[    7.842914]  __cmpxchg_case_acq_64+0x8/0x28
[    7.842922]  epoll_mutex_lock.constprop.0+0x18/0x3c
[    7.842938]  do_epoll_ctl+0x114/0x488
[    7.842947]  __arm64_sys_epoll_ctl+0x5c/0xc8
[    7.842952]  invoke_syscall+0x90/0x128
[    7.842961]  el0_svc_common.constprop.0+0xc8/0x114
[    7.842968]  do_el0_svc+0x38/0x4c
[    7.842973]  el0_svc+0x30/0x5c
[    7.842978]  el0t_64_sync_handler+0xb0/0x144
[    7.842984]  el0t_64_sync+0x1a4/0x1a8
```

KlimRider, 2024-05-03:
Yes I use stock power supply from Orange pi.
Exactly 5v 4A

Joshua-Riek, 2024-05-03:
> Yes I use stock power supply from Orange pi. Exactly 5v 4A

The stock power supply I received with my Orange Pi 5 Pro was a PD power supply that was not 5V 5A. Can you confirm this charger is 5V 4A?

Joshua-Riek, 2024-05-03:
Right, so it's probably DDR blob and SPL problem. However I do see some issues related to GIC, can you add `irqchip.gicv3_pseudo_nmi=0` to the kernel command line in `/boot/extlinux/extlinux.conf`?

KlimRider, 2024-05-03:
With this parameter at cold start was booting to the splash, then stuck
After reset kernel panic again
[Uploading irqchip.gicv3_pseudo_nmi_0.log…]()

KlimRider, 2024-05-03:
For now check eMMC module.
Flash on them image for opi 5 plus and install to plus.
Boot ok, work fine

Joshua-Riek, 2024-05-03:
> With this parameter at cold start was booting to the splash, then stuck After reset kernel panic again [irqchip.gicv3_pseudo_nmi_0.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15202392/irqchip.gicv3_pseudo_nmi_0.log)

The log output is mangled here, i cant read it.

Joshua-Riek, 2024-05-03:
> > With this parameter at cold start was booting to the splash, then stuck After reset kernel panic again [irqchip.gicv3_pseudo_nmi_0.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15202392/irqchip.gicv3_pseudo_nmi_0.log)
> 
> The log output is mangled here, i cant read it.

Nevermind i can read it, my editor was acting strange.

KlimRider, 2024-05-03:
No, server image same behavior

lp1964, 2024-05-04:
Hi
First of all, congratulations Joshua for the excelente job!.
As for the problem here reported (OPI 5 Pro not booting), the only way to have a boot and "normal" work with Ubuntu, is previously fash the OPI OS into eMMC and then use a SD card with the Ubuntu (in this case the OPI 5 boots and works only from the SD). No other combination works (using SD and/or eMMC). I'm extracting the debug logs in two situations (Ubuntu in SD & OPI OS in eMMC and Ubunti in SD & clean eMMC) and try to see the differences. Hope can help.
Best

KlimRider, 2024-05-05:
lp1964 thanks for the answer
Does this apply to the entire OPI5 line?
Did you manage to run this firmware on OPI5 Pro?

P.S. It is not entirely clear how the presence of an empty memory module or flashed with OPI OS and successful loading of kernel 6 relate to each other.
So the official firmware is installed on the eMMC memory module and works successfully. Even without an SD card. But with 5 kernel ony as OPI very lousy support In SW case

Joshua-Riek, 2024-05-05:
I'm going to need to bisect the opi kernel and my own. I really don't want to though as it takes a lot of time.

KlimRider, 2024-05-05:
If you can please describe what you want rdo. I Will try to do it by myself. I've got vacation from tomorrow.

lp1964, 2024-05-06:
Hi Joshua
Sorry for the delay in answer, but I've been away for the last two days...
In addition to the OPi5Pro, I have an OPi3B but I haven't tested it yet with kernel 6 (I'll do that in the next few days); I'm also waiting for the arrival of an OPi5Plus that I recently bought and will also test...
Regarding the OPi5Pro, here are the tests I did (I still have to "separate" the log files and I can then upload them); to be clear I call OPi OS (kernel 5); Ubuntu ("your" OS kernel 6):
OPi OS in SD (anything in eMMC, even nothing): boots and run the OPi OS
Ubuntu on SD (OPi OS on eMMC): boots and run Ubuntu
Ubuntu in SD (eMMC clean): not boots
Ubuntu in SD (Ubuntu in eMMC): not boots
Without SD (OPi OS in eMMC): boots and run the OPi OS
Without SD (Ubuntu in eMMC): not boots
Note that I kept all the options indicated for the OPi, namely Install boot in SD/eMMC; installing Ununtu boot (SD ou eMMC as inidcated), it doesn't work.
I didn't test with NVMe support as the SBC OPi5Pro out of the box doesn't have spi memory.
I'll uploading the debug logs...
Best (and good vacations)
Luiz

KlimRider, 2024-05-06:
Found strange behavior. If I start board by command reboot then my built-in wifi adapter is disappear. If I restart by hard reset or by turn off power than wifi adapter is alive and working. But I prefer think about this like feature not bug.

Joshua-Riek, 2024-05-06:
The wifi and Bluetooth will be kinda broken. It require sdio driver, but pcie is in use, and they are not compatible under the same kernel.

KlimRider, 2024-05-06:
They work and very well, but only after hard reset. Initially I think this bug cause already used PCIE slot. But no, after power reset work Wifi, Bluetooth, I think some garbage data in chip not cleaning during regular reboot. Broadcom bug (all they touched magically become brown substance).

salier, 2024-05-07:
> Hi Folks Today i try to flash new one image from the latest build to the eMMC card and
> 
> > Server image is booting well
> 
> * NVME (ZFS) is working
> * ETH is working
> * WIFI is working
> * NPU in dmesg is shown but not tested yet
> 
> > Desktop image is booting well too
> 
> * NVME (ZFS) is working
> * ETH is working
> * WIFI is working
> * NPU in dmesg is shown but not tested yet
> * Graphics Accelleration - glmark2 shown GA is active, but chromium not start on first run (EGL Error). Then start OK and accelerated
> * Sound play work well from both HDMI (analog out/in too)
> 
> Joshua Thank you again

Hello, which img is being used for this? I see that the latest img has not been updated.

KlimRider, 2024-05-07:
Hi
In git you can see activity of owner of project.
https://github.com/Joshua-Riek/ubuntu-rockchip/actions
Not all builds mark as release 
As I remember I test image from this build - https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/8957854021
But you may want to test the latest build.

salier, 2024-05-07:
Indeed, I do want to test the latest version of the image.

salier, 2024-05-07:
I tried ubuntu-22.04 pre installed desktop arm64 rangepi-5 pro, and it should look consistent with the 6.1 kernel provided by Orangepi. After installation, although the LED lights display and start normally, SSH cannot be performed and the screen cannot be seen.

Joshua-Riek, 2024-05-11:
I updated the DDR and SPL blobs again and the Orange Pi 5 Pro is now stable. I hope to have a release out this weekend with the latest changes.

closed 2024-05-11 by Joshua-Riek

lp1964, 2024-05-12:
I tried 22.04 desktop build 475 (for some unknown reason I couldn't download the latest one). I also have the 24.04 but didn't have time to test.
Tested on OPi5 Pro, flashed on eMMC. Boots up, configure and works perfectly! I didn't have time to test the audio; the GPU seems to work perfectly well as the wireless module (wifi and BT). All previously referred problems appear to be corrected.
I will update the information as more testing is done.
Great work Joshua!