Joshua-Riek/ubuntu-rockchip #769: I2C communication error, unable to start 24.04 LTS on orangepi 5.

Quantumobserver, 2024-05-03:
I tried to start the latest ubuntu24.4 image in orange pi5 but it did not start successfully. By looking at the startup log, I found that it was stuck in the initialization of I2C communication.
The following is the startup log:
[bootlog.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15203688/bootlog.txt)

Joshua-Riek, 2024-05-03:
Is this Ubuntu 24.04 server? If so I don't see any issue, the `Goodix-TS` is a display panel and since there is likely no display panel connected to your devices this error is expected and would not prevent the system from starting. 

Wait about one or two minutes and the system should enter the login screen.

Quantumobserver, 2024-05-03:
Thanks for your help, I noticed that my urat serial adapter is not working. It should be able to access the command line directly, I tried it in recovery mode and it worked.
Happily, I waited for a while and it made its way into the system.
Thank you again for your great contribution!

closed 2024-05-03 by Quantumobserver