Joshua-Riek/ubuntu-rockchip #773: Gstreamer-rockchip mpp encoders issue

Xeue, 2024-05-05:
I have been testing using the gstreamer mpph264enc and mpph265enc elements.
I am seeing slow encoding performance, when I first run a pipeline it will play normally for a few seconds sometimes, then freeze, then occasional frames will be rendered. There's can be many seconds between a single frame being rendered.
I tested using ffmpeg and saw no issues at all.

I will get full logs tomorrow when I am back in, but I didn't see anything jump out in gstreamer logs relating to the encoder.

I was running commands like:
`gst-launch-1.0 videotestsrc ! video/x-raw,height=1080,width=1920,framerate=50/1 ! videoconvert ! mpph265enc ! fakesink`

You will see the counter in command line freezing and playing randomly.

I tried many combinations of options on the encoder, feeding many different video sources aside from the videotestsrc, with different pixel formats etc. I have also tried many destinations, from SRT to fakesink.

I have tested on a completely fresh install of the 2.0 release with my first commands being the gstreamer pipelines. I was testing on the NanoPC T6 LTS.

I will do more investigation and log collection tomorrow!

nyanmisaka, 2024-05-06:
There is a regression in previous commits of `gstreamer-rockchip->mpph26xenc`. https://github.com/JeffyCN/rockchip_mirrors/issues/46

Xeue, 2024-05-06:
Thanks for highlighting that!
Are there any instructions anywhere on how I can compile/build that and test it?

If not, @Joshua-Riek when you have some spare time (I know you're very busy and have many other things to look at!) could you get the fixed version in your PPA?

Thank you both for all you do!

Xeue, 2024-05-07:
Just built the package with the following steps:
git clone --branch gstreamer-rockchip https://github.com/JeffyCN/rockchip_mirrors.git --depth=1
cd rockchip_mirrors/
mkdir build
meson build
cd build
ninja
ninja install

And I still have the same issue

Here's the gstreamer output for this pipeline: GST_DEBUG=6 gst-launch-1.0 videotestsrc ! mpph265enc ! fakesink &> testsrctest.log
[testsrctest.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15239744/testsrctest.log)

Joshua-Riek, 2024-05-07:
> Just built the package with the following steps: git clone --branch gstreamer-rockchip https://github.com/JeffyCN/rockchip_mirrors.git --depth=1 cd rockchip_mirrors/ mkdir build meson build cd build [...]

The build process on ubuntu is a little diffrent.
```
ubuntu@ubuntu-desktop:~/rockchip_mirrors$ sudo apt build-dep .
ubuntu@ubuntu-desktop:~/rockchip_mirrors$ fakeroot dpkg-buildpackage -us -uc
```

I would try the above to compile the package, but i will start to upload the latest rockchip gstreamer for noble.

Joshua-Riek, 2024-05-11:
Any further problems after updating?

Xeue, 2024-05-11:
I've been testing for the last few hours.
All looks good!

closed 2024-05-11 by Joshua-Riek