Joshua-Riek/ubuntu-rockchip #774: rock5b ubuntu 24.04  server Install system to eMMC fails to boot up

kylin7226, 2024-05-06:
rock5b ubuntu 24.04  server Install system to eMMC fails to boot up

According to the official documentation of Rock5b, I installed the image onto the eMMC storage, but now it fails to boot up.

I use this image [ubuntu-24.04-preinstalled-server-arm64-rock-5b.img.xz](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.0.0/ubuntu-24.04-preinstalled-server-arm64-rock-5b.img.xz)

https://docs.radxa.com/en/rock5/rock5b/low-level-dev/install-os-on-emmc-from-usb-otg

Ukhupacha, 2024-05-06:
I guess you would have to install uboot on the SPI.

Joshua-Riek, 2024-05-07:
Don't follow the Radxa docs, follow what is on the Wiki please, i tested this on my end and it works as expected.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS

closed 2024-05-11 by Joshua-Riek