Joshua-Riek/ubuntu-rockchip #777: Orange Pi 5 Plus Ubuntu 24.04 Frigate (13.0.2-rk)

flyzet-prog, 2024-05-07:
I have an issue with Frigate on Kernel 5.1. I got green records, when i try to restart system it freezing. I have to restart it with button. Maybe some issues with RKNN version?
[image]

Joshua-Riek, 2024-05-07:
Ubuntu 24.04 does not use kernel 5.10?

flyzet-prog, 2024-05-07:
Sorry, maybe i something missing?
[image]

flyzet-prog, 2024-05-07:
I use this one that on screenshot, from 2024-01-30. Seem like v 1.33.
[image]

Joshua-Riek, 2024-05-07:
The issue title says Ubuntu 24.04?

flyzet-prog, 2024-05-07:
> The issue title says Ubuntu 24.04?

Sorry, i don't understand. You mean it's 24.04.3?

Joshua-Riek, 2024-05-07:
You opened the issue "Orange Pi 5 Plus Ubuntu 24.04 Frigate (13.0.2-rk) - green records", this implies there is a problem on Ubuntu 24.04. But you show that you are running Ubuntu 22.04.

flyzet-prog, 2024-05-07:
> pened the

Sorry, my bad.

closed 2025-01-04 by flyzet-prog