Joshua-Riek/ubuntu-rockchip #779: Compilation error, armsom-sige7

armsom-team, 2024-05-07:
[image]

Joshua-Riek, 2024-05-07:
Hey! I'm now using livecd-rootfs from upstream Ubuntu and sadly it can easily break. You need to use an up-to-date Ubuntu 24.04 or Ubuntu 22.04 system. Right now, the easiest way to build is by forking the project and building with GitHub actions. 

I recently try to use docker to reduce the number of build issues, however it introduces a dbus issue on Ubuntu 24.04 generated images where the desktop installer will crash.

hakotani-o, 2024-05-08:
Thank you for greate work and great build system.

I try build ubuntu on Orange pi 5.

kernel 0012 error stop in live-build.
[image]

kernel 0011 live-build finished.
[image]

please try kernel 0011.

Thank you.

Joshua-Riek, 2024-05-08:
I disabled Apparmor by default due to a lot of errors on startup, let me take a look at livecd-rootfs and see if i can workaround this problem.

Joshua-Riek, 2024-05-08:
I disabled a few references to apparmor as a quick test, https://github.com/Joshua-Riek/livecd-rootfs/commit/a2de77e48914b25bd6480c29149a13549029239b, please try to build again and hopefully it will work. I will also test on my system.

Joshua-Riek, 2024-05-08:
I may need to enable apparmor again in the kernel, for some reason it needs to be enabled to preseed snaps.. 

```
Fetching assertions for "lxd"
Install the snap with:
   snap ack lxd_28474.assert
   snap install lxd_28474.snap
renamed 'chroot/var/lib/snapd/seed/lxd_28474.assert' -> 'chroot/var/lib/snapd/seed/assertions/lxd_28474.assert'
renamed 'chroot/var/lib/snapd/seed/lxd_28474.snap' -> 'chroot/var/lib/snapd/seed/snaps/lxd_28474.snap'
ubuntu classic preseeding
unmounting: /home/ubuntu/ubuntu-rockchip/build/ubuntu-live-build/build/chroot/tmp/snapd-preseed
ubuntu classic preseeding
unmounting: /tmp/snapd-preseed
error: cannot preseed without access to "/home/ubuntu/ubuntu-rockchip/build/ubuntu-live-build/build/chroot/sys/kernel/security/apparmor"
```

hakotani-o, 2024-05-08:
Hi Joshua-Riek.
Yes.
Deployment of snap package fails on kernel 0012.
That's why I'm building with kernel 0011.

Sorry, I have to go out now, so I'll try to build it when I get back.
Thank you.

hakotani-o, 2024-05-08:
Hi Joshua-Riek.
I try build again.
live-build fail snap install.
Thank you.

hakotani-o, 2024-05-08:
Hi Joshua-Riek.
Thank you for new kernel.
With the latest kernel, the live build is now able to download and extract snap packages.
The device can also be unmounted at the end of the live build.

Thank you.

closed 2024-05-22 by Joshua-Riek