Joshua-Riek/ubuntu-rockchip #780: Ubuntu 24.04 won't boot from SATA SSD on Orange Pi 5

colornoise42, 2024-05-07:
1. Booted Ubuntu 24.04 LTS Release from SD card
2. Added to /etc/default/u-boot
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/orangepi-5-sata.dtbo"
3. sudo u-boot-update 
reboot
4.  Install U-Boot to the SPI Flash:
sudo flashcp /usr/lib/u-boot/rkspi_loader_sata.img /dev/mtd0
5. Install Ubuntu onto an SATA SSD from Linux:
sudo ubuntu-rockchip-install /dev/sda
poweroff
6. Pulled out the SD card
power on - the system does not boot
<details><summary>boot log:</summary>
<p>

```shell
DDR V1.11 f1474cf52f cym 23/05/09-11:02:36
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x1
CH0 RX Vref:27.9%, TX Vref:20.8%,20.8%
CH1 RX Vref:28.5%, TX Vref:22.8%,22.8%
CH2 RX Vref:31.0%, TX Vref:20.8%,20.8%
CH3 RX Vref:30.1%, TX Vref:23.8%,22.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09 (May 05 2024 - 16:19:14)
Trying to boot from MMC1
spl: mmc init failed with error: -123
Trying to boot from MMC2
Card did not respond to voltage select!
spl: mmc init failed with error: -95
Trying to boot from MTD2
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(7efcd01a0f...) + OK
## Checking uboot 0x00200000 ... sha256(3ebd5db017...) + OK
## Checking fdt 0x0034a300 ... sha256(60482be298...) + OK
## Checking atf-2 0xff100000 ... sha256(1163474a5b...) + OK
## Checking atf-3 0x000f0000 ... sha256(da90adf3a4...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 504.768 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-589-g3389cfdda:derrick.huang
NOTICE:  BL31: Built : 10:14:29, May  9 2023
INFO:    spec: 0x13
INFO:    ext 32k is valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
[... 231 lines cut ...]
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0  33.7M  1 loop /snap/snapd/21467
loop1         7:1    0  69.2M  1 loop /snap/core22/1383
loop2         7:2    0  93.7M  1 loop /snap/lxd/28474
loop3         7:3    0  92.8M  1 loop /snap/lxd/28475
sda           8:0    0 238.5G  0 disk
└─sda1        8:1    0 238.5G  0 part
mtdblock0    31:0    0    16M  0 disk
mmcblk1     179:0    0 119.4G  0 disk
└─mmcblk1p1 179:1    0 119.4G  0 part /
``` 

</p>
</details> 

<details><summary>blkid</summary>
<p>

```shell
/dev/mmcblk1p1: LABEL="desktop-rootfs" UUID="b7e2aefb-e1b1-4739-b8c5-fe1403237b15" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="78f64f4c-aecf-4b86-9bbf-95d04d174604"
/dev/sda1: LABEL="desktop-rootfs" UUID="a767c44e-afb0-4d09-8b71-47aba39a4921" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="4764849b-c148-43c6-a0cc-ca15ce6b0fcb"
``` 

</p>
</details>

Joshua-Riek, 2024-05-07:
I see the issue, the system is trying to load rk3588s-orangepi-5-sata.dtb which does not exist.

Joshua-Riek, 2024-05-11:
Update the kernel and this will be fixed.

closed 2024-05-11 by Joshua-Riek

colornoise42, 2024-05-14:
Thank you! Updating the kernel and u-boot solved the problem.