Joshua-Riek/ubuntu-rockchip #782: [24.04] Radxa A8 Module Issues

MyVizDrake, 2024-05-07:
Originally from [here](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/727) during the Beta
>I tried sudo modprobe -r btusb .... sudo modprobe btusb and just tried sudo modprobe btusb again at your direction .... nothing is showing up in the Gnome Bluetooh settings window. It's not that it can't see the two devices I care about, it also can't see all of the other USB devices that Android image could see. That being said, my ArchLinux install that actually has a working DSI display can't see any BT devices either ... I even turned BT off and back on again ..

Device seems to be available.

[image]

From [Kwankiu's ArchLinux-Installer](https://github.com/kwankiu/archlinux-installer/wiki/General#troubleshooting)
>Note: it seems like there may be issues with bluetooth on linux-radxa-rkbsp5-git (5.10.160), consider using linux-rockchip-joshua (5.10.160) if you need bluetooth.

 I am fairly certain I have the Radxa 5.10.160 kernel on that one (which so far is the only one I can get my Radxa 10FHD Display to work).

No rush. You're doing great work!

Joshua-Riek, 2024-05-08:
I backported RTW89 from Linux 6.1 to provide improved performance and compatibility https://github.com/Joshua-Riek/linux-rockchip/commit/8285966d2a5693c7b461f30d3e822d06a18f7342. I will be honest that I have no intent to debug the wireless driver at this time, too much effort on my part.

MyVizDrake, 2024-05-08:
Understood ... :-)

MyVizDrake, 2024-05-13:
@Joshua-Riek closing this ... once I blacklisted all of these modules in blacklist.conf instead of blacklist it works.

> blacklist pgdrv
> blacklist btusb
> blacklist btrtl
> blacklist btbcm
> blacklist btintel

Fell victim to autocomplete and didn't notice it didn't do the complete file name.
On to my DSI display.

Thanks again!

closed 2024-05-13 by MyVizDrake