Joshua-Riek/ubuntu-rockchip #784: Orange pi 5 pro boot from usb

gsweeh, 2024-05-09:
I just bought the OP5 Pro. I've just realized that there is no SPI flash fitted on the board. I simply want to boot from an external HDD. I've also seen the manual for booting from USB, but I believe it requires an SPI flash or eMMC module. Is there any way to boot from USB without either of these components?

KaMyKaSii, 2024-05-14:
You can try flashing the bootloader image to an SD card and see if it makes the OP5 Pro load the OS from the USB disk. The attached bootloader image was extracted from the "ubuntu-24.04-preinstalled-desktop-arm64-orangepi-3b.img.xz" image

[u-boot-rockchip-spi.bin.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15310738/u-boot-rockchip-spi.bin.zip)

gsweeh, 2024-05-14:
I want to boot from external hard disk (usb).

KaMyKaSii, 2024-05-14:
> I want to boot from external hard disk (usb).

I got it. But for this you need to have the bootloader installed on a disk where the board searches and then this bootloader will identify the OS on the USB disk and boot it. Anyway, I took a look and from what I saw the bootloader image needed to be installed on the SD card is different from the SPI one, I'll attach the correct one. Basically install the OS to your USB device using BalenaEtcher or similar and also install the attached image on an SD card using the same method. Note that to start the OS, the SD card with the bootloader will always need to be connected to the board in the boot time. 
[u-boot-rockchip.bin.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15311792/u-boot-rockchip.bin.zip)

gsweeh, 2024-05-15:
Sorry, I'm new to this Orange Pi world. First, do I need to flash [ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5-pro.img.xz] to a USB external drive via BalenaEtcher and [u-boot-rockchip.bin.zip] to an SD card via same method, right?

KaMyKaSii, 2024-05-15:
> Sorry, I'm new to this Orange Pi world. First, do I need to flash [ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5-pro.img.xz] to a USB external drive via BalenaEtcher and [u-boot-rockchip.bin.zip] to an SD card via same method, right?

Yes. Then plug them both into the board and turn it on. You are using the latest release (v2.1.0), right?

gsweeh, 2024-05-15:
(v2.1.0) for which software?

gsweeh, 2024-05-15:
it is not working after flashing in sd card

FreeWilly2022, 2024-05-16:
I attempted to do what KaMyKaSii suggested, except with the system on the NVMe SSD and it was not successful. I have 24.04 V2.1 desktop running from an SD card on my 16 GB OPi5 Pro, and then used the sudo ubuntu-rockchip-install /dev/nvme0n1 command to copy it to my SSD. Since I do not have SPI flash or eMMC I need to put the bootloader onto an SD card to run off the SSD. I used balenaEtcher to write KaMyKaSii's u-boot-rockchip.bin.zip file per the above post (after unzipping it) onto a second SD card, inserted it into the OPi5Pro, and it doesn't boot. So I connected a USB serial converter to the serial debugging port. I confirmed that it works by booting from the SD card with the OS on it, and watched the system boot up (I had to disable BRLTTY). Then I put the SD card with just the bootloader in, and nothing appears on the serial debugging port when I try to start it, and the system doesn't boot at all. Joshua has a command to write U-boot to the SD card, but that doesn't help when the system is running off the SD card (I tried it anyway, but as soon as I pulled out the SD card with the OS ubuntu started puking out errors because it needed the card, and it didn't work).

I think other folks will get the Pro without eMMC and not want to solder in the SPI flash, so it would be a nice problem to solve. But I don't even know where to start, so any suggestions would be appreciated.

KaMyKaSii, 2024-05-17:
So I took a look at u-boot-install and it skips 32 kilobytes in the device block before writing the u-boot-rockchip.bin file, maybe that's why it's not working. In the attached file there is the u-boot-32k-gap.bin file where I add the blank bytes that are skipped in case you want to try installing through BalenaEtcher again, or you can use the modified script u-boot-install.sh to install directly on the SD card with the command "./u-boot-install.sh /dev/mmcblk0" (replace the block device path if necessary)
[u-boot-sdcard.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15342650/u-boot-sdcard.zip)

gsweeh, 2024-05-17:
@FreeWilly2022, bro, did you try this? Is it working for you?

gsweeh, 2024-05-17:
@KaMyKaSii 
[image]
, I tried your new comment guide. First, I installed the latest version of Ubuntu 22 on an SD card and also on an external hard drive. Then, I inserted the SD card into the board and it booted with the initial setup. After that, I flashed the bootloader on the SD card using:

```
./u-boot-install.sh /dev/mmcblk1p1
```

If I execute the `lsblk` command, my external hard disk shows up as `sda1`. After flashing the bootloader, I restarted the board. I connected both the external hard disk and the SD card. The output will be as follows. I have attached the image.

gsweeh, 2024-05-17:
@KaMyKaSii @FreeWilly2022 Guys, it is working now. I can boot from USB. Thank you so much!
[image]

gsweeh, 2024-05-17:
After i restarted board. it give this log. It can solve?
[image]

FreeWilly2022, 2024-05-17:
KaMyKaSii thanks for helping! Unfortunately I tried this both with BalenaEtcher and with the bash script and I got nothing on the serial debugging port. Since gsweeh was able to get it to boot, I must be doing something wrong, and will keep trying.

FreeWilly2022, 2024-05-17:
gsweeh - you said you used ./u-boot-install.sh /dev/mmcblk1p1 which is actually writing to a partition, not a device. So I tried writing the bootloader on the SD card using that command and it gave me an error that it was not a device. Are you sure this is the command you used? Did you write to the same SD card that you booted off of?

KaMyKaSii, 2024-05-17:
> @FreeWilly2022 I follow his new comment guide. First, I installed the latest version of Ubuntu 22 on an SD card and also on an external hard drive. Then, I inserted the SD card into the board and it booted with the initial setup. After that, I flashed the bootloader on the SD card using:
> 
> ```
> ./u-boot-install.sh /dev/mmcblk1p1
> ```
> 
> If I execute the `lsblk` command, my external hard disk shows up as `sda1`. After flashing the bootloader, I restarted the board. I connected both the external hard disk and the SD card. For the first time ubuntu booted from external hard disk then i did initial setup. After i restarted a board then it shows output like this ![IMG_0407](https://private-user-images.githubusercontent.com/65717972/331577805-94c67943-1786-4251-aaa3-c09009025276.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU5Njc1ODYsIm5iZiI6MTcxNTk2NzI4NiwicGF0aCI6Ii82NTcxNzk3Mi8zMzE1Nzc4MDUtOTRjNjc5NDMtMTc4Ni00MjUxLWFhYTMtYzA5MDA5MDI1Mjc2LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxN1QxNzM0NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xODk2YzAwMWU3NmJiZTIyZmVkZmVjNWYyNGVkYmVjMWQ1ODljMzdiNzYxYzkwMWQ4ZmU3NDFhMGQyOGNmNDJlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BE6_fUvO5I6mwao3KiFHCSbHPf9kykLnL0zC-V9fGkQ)

You wrote the bootloader over the ext4 partition, that's why it got corrupted. Anyway, I took another look and I think the method in my last message is still not the correct one. I checked my current installation of Ubuntu Rockchip (I have an Orange Pie 3B) and verified that the bootloader installation seeks 32 kilobytes ahead of the device block so as not to overwrite the partition table, which is apparently necessary. 
So I want you two to try this: use a working computer to install the Ubuntu Rockchip image to both the USB disk and the SD card. Once complete, use GParted or another partition management program to delete "desktop-rootfs" from the SD Card only. Then eject and place both in the Orange Pie and turn it on. With luck, the bootloader on the SD Card will find the "desktop-rootfs" partition of the USB disk and mount it as the system root and then boot Ubuntu normally.

FreeWilly2022, 2024-05-17:
@KaMyKaSii thanks again for your help. I used GParted and deleted the desktop-rootfs partition. It showed the whole SD drive as unallocated. When I put the SD card in the OPi5 Pro and turned on the power, the desktop loaded immediately. This is a much cleaner way of making a boot only SD card.

gsweeh, 2024-05-18:
Finally, it is working. Could you guys tell me how I can install RDP or VNC on the board? Whenever I try to install it myself, it shows a blank screen. Is there a proper way to install it?

cantux-fr, 2024-05-18:
I have the same problem that gsweeh had. Can you summarize what needs to be done to boot from USB and the simplest way to do it?

gsweeh, 2024-05-18:
Just flash the image to both the SD card and USB hard drive using balenaEtcher. After flashing the image, use any partition management tool. I used MiniTool Partition Wizard on Windows. Then, delete "desktop-rootfs" from the SD card. Finally, connect the SD card and USB to the board, and start it. It will boot from the USB.

cantux-fr, 2024-05-18:
Just flash the image to the SD card and the USB hard drive using balenaEtcher. For the USB hard drive, I'll use the Ubuntu image, but which image should I use for the SD card?

FreeWilly2022, 2024-05-18:
@cantux-fr always use the same image for the SD card as you put on the USB, and yes, when you delete "desktop-rootfs", there are 0 partitions left. Its ok, just put the card in the SD drive and turn on the computer.

cantux-fr, 2024-05-18:
ok, thx a lot :-)

cantux-fr, 2024-05-18:
It doesn't work. Here's what I did:
I downloaded "ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5-pro.zip" from this page: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9036485442
I unzipped the .zip file.
I have a micro SD card and a USB NVME hard drive. I deleted all partitions on both using Gparted. I created an "ms-dos" partition table on both. And I created a partition that takes up all the space with the FAT32 format on both. I flashed the micro SD and the USB NVME with Balena Etcher using the "ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5-pro.img.xz" file. I used the partition editor to delete the "desktop-rootfs" partition on the micro SD. I plugged the micro SD card into the micro SD port of the Orange Pi 5 Pro. I plugged the USB NVME into a USB port of the Orange Pi 5 Pro. When I plug in the Orange Pi 5 Pro, it doesn't start (black screen). I tried with another partition editor "MiniTool Partition Wizard". I also tried connecting the USB NVME to other USB ports of the Orange Pi 5 Pro. But it still doesn't start. Did I miss a step? Thank you very much :-)

FreeWilly2022, 2024-05-18:
@cantux-fr - don't delete the partitions the USB drive, only the SD card. Don't create any partitions, the SD card will show 0 partitions, that is ok (I don't actually know why but I think its because the first stage boot on the ROM just reads the card without a file format).

Then put the SD card in and connect the USB drive, and power it up.

KaMyKaSii, 2024-05-18:
> It doesn't work. Here's what I did: I downloaded "ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5-pro.zip" from this page: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9036485442 I [...]

You do not need to previously format (create a new partition table) and create a partition taking all the space on the SD Card because the Ubuntu Rockchip image already contains a partition table that will overwrite the old one. The part about deleting the desktop-rootfs partition from the SD Card is necessary so that the bootloader does not get confused about which partition to boot from. In fact, I think that after deleting the SD Card partition it is possible to create a new partition with the filesystem you want to take advantage of the SD Card storage, just make sure NOT to activate any special flag on this new partition such as "EFI System". Returning to your case, is there any special reason why you are using images from the actions page and not the releases page? It would be interesting in addition to testing the releases page image, perhaps you would like to record yourself doing the procedure so we can see if you are doing something wrong

KaMyKaSii, 2024-05-18:
> Finally, it is working. Could you guys tell me how I can install RDP or VNC on the board? Whenever I try to install it myself, it shows a blank screen. Is there a proper way to install it?

I don't have much knowledge in remote management but I know there are some problems with this part when running the system on Wayland (which is the default), try searching for "Ubuntu Wayland VNC" to see if there is any solution to your problem. You may have to switch from Wayland to Xorg to be able to use remote management correctly. In my case, I managed to connect from my desktop PC to the board (Ubuntu Rockchip 24.04 on Orange Pi 3B) using the RustDesk app, even using Wayland, but the system was very slow, I imagine it is because RustDesk is not able to use the GPU hardware acceleration to encode the video. Note that if you opt for a solution like this, if you don't have a real monitor connected to the board, in order to have video when connecting remotely you may need to have a dumb video output connected to the board, I know there are solutions of hardware for this but perhaps there is software too

FreeWilly2022, 2024-05-18:
To try to help @cantux-fr, and @KaMyKaSii may be interested, I removed my NVMe card from the M.2 slot, plugged in a USB adapter with the SD card that has the system (24.04 desktop V2.1), and started the OPi5 Pro. I have the serial console and I could see it boot and look for USB devices, but it reported 0 Storage Devices found. I tried a second USB adapter and it didn't work either. I do not have a USB drive large enough to try, so that is the best that I can do. @cantux-fr if you can, you might want to try putting the NVMe into the M.2 slot and see if that works. Otherwise, it may be that some USB devices are not recognized as storage devices by the U-boot?

cantux-fr, 2024-05-18:
I put the NVME M2 in the slot of the Orange Pi 5 Pro and disconnected the micro SD card, but it still doesn't work.

FreeWilly2022, 2024-05-18:
Did you re-write all the images? You should start from the beginning, flash the image to the SD card, then startup and go through the initial installation. 
Then open a terminal and do the command described in the wiki https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS sudo ubuntu-rockchip-install /dev/nvme0n1
Then reboot using the SD card with the "desktop-rootfs" partition deleted.

cantux-fr, 2024-05-18:
Can you describe to me in detail how to proceed? I'm having trouble with the tutorial using the terminal.

FreeWilly2022, 2024-05-18:
Start the Terminal program, it will open a window with a $ prompt. Type:
 sudo ubuntu-rockchip-install /dev/nvme0n1
press return, and it will ask for your password. Type your password and press enter, and it will write the OS onto the SSD.

cantux-fr, 2024-05-18:
With this command, it gives me:
ubuntu-rockchip-install: command not found

FreeWilly2022, 2024-05-18:
Unfortunately I don't know enough Linux to figure out what is wrong. Maybe someone else can help? You can also try asking on Reddit r/OrangePi, that is more of a chat format.

cantux-fr, 2024-05-20:
When I plug in my NVME USB hard drive, it is mounted under the name /dev/sda1
I run the following command.
$ sudo ubuntu-rockchip-iunstall /dev/sda1
[sudo] password for cantux:
This script will install the currently running system onto /dev/sda1.
It may take upto a few minutes - Continue? [y/N] y
Creating partition table for /dev/sda1.
Error: in /usr/bin/ubuntu-rockchip-install on line 57

Tekunalogy, 2024-05-20:
> Just flash the image to both the SD card and USB hard drive using balenaEtcher. After flashing the image, use any partition management tool. I used MiniTool Partition Wizard on Windows. Then, delete [...]

Thank you for this. I added instructions that worked for my RockPi 5 Pro: https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/821

serialscriptr, 2024-10-07:
Had to do this to get nvme booting to work on the orange pi 5 pro, wiki instructions here do not work

closed 2024-12-16 by github-actions[bot]