Joshua-Riek/ubuntu-rockchip #785: Override cloud-init

jsantosa-minsait, 2024-05-09:
Hi,

just a simple question. There is a way to override cloud-init configuration instead using the default `overlay/boot/firmware` without rebuilding the whole image. Just for the first startup.

I have tried to copy user-data, network-config and meta-data files to `/mnt/boot/firmware` after copying the imege to the nvme drive but those files are override with `overlay/boot/firmware`.

Thanks in advance

Best regards,
Javier

nilo85, 2024-05-09:
Is it server image? There is a fat32 partition where you can replace the files and they are used at first boot

jsantosa-minsait, 2024-05-09:
> Is it server image? There is a fat32 partition where you can replace the files and they are used at first boot.
> 
> works for me, but not tried booting fork nvme, booting from emmc but suspect it should be universal

Thanks @nilo85 

Haven't seen that partition when flashing the image... and yer, it's the server one.

nilo85, 2024-05-10:
I think it’s only there on the 24.04 version though, so it could be it’s missing on the 22.04 if that is what you are using

jsantosa-minsait, 2024-05-10:
> I think it’s only there on the 24.04 version though, so it could be it’s missing on the 22.04 if that is what you are using

Thanks a lot @nilo85 
It works just fine on both distributions!! 

Regards
Javier

closed 2024-05-10 by jsantosa-minsait