Joshua-Riek/ubuntu-rockchip #786: Ubuntu 24.04 (Orange Pi5B) Error show cpu (neofetch)

ArtemS2, 2024-05-10:
In the beta version there was a display of processor cores, but in the release version it disappeared somewhere. (neofetch)
[image]

cantalupo555, 2024-05-11:
Orange Pi 5 👇
[image]

Joshua-Riek, 2024-05-11:
I see that python3 is not in the dependency list of Neofetch, it's required for multiple CPU detection. 

Thanks for reporting this, I've updated the package, it may take a few hours to be downloadable via apt.

https://launchpadlibrarian.net/729343706/neofetch_7.1.0-4+2_7.1.0-4+3.diff.gz

closed 2024-05-11 by Joshua-Riek