Joshua-Riek/ubuntu-rockchip #787: Errors when turning off. Ubuntu 24.04 Orange PI5B

ArtemS2, 2024-05-10:
Errors when turning off.  This doesn’t seem to be critical, but it’s annoying.

[image]

Joshua-Riek, 2024-05-11:
Does this cause an actual problem such as the system not being able to power down??

ArtemS2, 2024-05-11:
There are no particular problems with this, I was just confused by the error indication. The system turns on quickly when needed.

Joshua-Riek, 2024-05-11:
Since it does not cause any problems, I likely will not look into this. 

Because the system is shutting down it starts to unload multiple kernel drivers including bluetooth and wifi, this is most likely why you are seeing this message.

closed 2024-05-11 by Joshua-Riek