Joshua-Riek/ubuntu-rockchip #789: How to use Panfrost on RK3588?

eternal-Eason, 2024-05-11:
Hello, I just compiled the root file system and put it on my RK3588 development board, but I found that Panfrost is not being utilized. How can I enable it? Thank you.

Joshua-Riek, 2024-05-11:
The panfost driver is a joke for RK3588, it requires a hostile user-space hack named Panfork to be some-what usable. I have no idea how you compiled the root filesystem or what tweaks were made for your specific system.

eternal-Eason, 2024-05-11:
@Joshua-Riek 

The latest code.

My compilation command is: sudo ./build.sh --board=rock-5b --release=jammy --project=desktop.

After obtaining the root file and placing it on my RK3588, upon system startup, I found that Panfrost did not start.

Do I need to change anything in the kernel configuration?

Thank you.

eternal-Eason, 2024-05-11:
> @Joshua-Riek
>
> The latest code.
>
> My compilation command is: sudo ./build.sh --board=rock-5b --release=jammy --project=desktop.
>
> After obtaining the root file and placing it on my RK3588, upon system [...]

2024/5/11 11:50:25

I just want to enable 3D hardware acceleration support via Panfrost.

Joshua-Riek, 2024-05-11:
3D hardware acceleration works out of the box. How have you confirmed that the GPU is not being used?

eternal-Eason, 2024-05-11:
[image]

In glmark2, the GPU is not utilized; instead, CPU rendering is used.

eternal-Eason, 2024-05-11:
After recompiling the root filesystem, I managed to enable the Panfork driver. However, GDM defaults to using the X11 protocol, which causes laggy performance in Chromium. When I use sudo systemctl restart gdm, it switches to the Wayland protocol, resulting in smoother performance in Chromium. Sometimes, upon device restart, it defaults to X11, while other times it defaults to Wayland. How can I ensure that it consistently uses Wayland? Thank you.
[image]

[image]

aubreyrjones, 2024-05-11:
@eternal-Eason You can edit `/etc/gdm3/custom.conf` and add `XorgEnable=false` under the `[daemon]` section. That _should_ work. The opposite (disabling wayland) certainly works.

eternal-Eason, 2024-05-14:
> @eternal-Eason You can edit `/etc/gdm3/custom.conf` and add `XorgEnable=false` under the `[daemon]` section. That _should_ work. The opposite (disabling wayland) certainly works.

I reverted to version 1.30 and found that I can successfully use Wayland without encountering the phrase  X11。 Thank you.

closed 2024-05-14 by eternal-Eason

wanyingwe, 2024-09-24:
您好，我查找3588使用panfrost的方法，基本都是基于wayland系统，而我的系统是X11的，使用相同的安装方式没有能成功，请教一下您是如何成功在X11的系统上安装的panfrost，有什么需要特别的步骤么？

eternal-Eason, 2024-09-24:
之前我的板子多屏显示的，我猜测系统没有做好多屏显示，开机一直是是X11，后面只用一个屏幕就是wayland ，使用X11会有一个问题：使用chromium特别卡。使用x11一般是/etc/gdm3/custom.conf and add XorgEnable=true。这个不确定。armbian系统可以开机时候设置选择x11或者wayland。armbian可以解决你的问题，他们也有Ubuntu系统。

eternal-Eason, 2024-09-24:
> 您好，我查找3588使用panfrost的方法，基本都是基于wayland系统，而我的系统是X11的，使用相同的安装方式没有能成功，请教一下您是如何成功在X11的系统上安装的panfrost，有什么需要特别的步骤么？

之前我的板子多屏显示的，我猜测系统没有做好多屏显示，开机一直是是X11，后面只用一个屏幕就是wayland ，使用X11会有一个问题：使用chromium特别卡。使用x11一般是/etc/gdm3/custom.conf and add XorgEnable=true。这个不确定。armbian系统可以开机时候设置选择x11或者wayland。armbian可以解决你的问题，他们也有Ubuntu系统。

eternal-Eason, 2024-09-24:
@wanyingwe  之前我的板子多屏显示的，我猜测系统没有做好多屏显示，开机一直是是X11，后面只用一个屏幕就是wayland ，使用X11会有一个问题：使用chromium特别卡。使用x11一般是/etc/gdm3/custom.conf and add XorgEnable=true。这个不确定。armbian系统可以开机时候设置选择x11或者wayland。armbian可以解决你的问题，他们也有Ubuntu系统。

wanyingwe, 2024-09-24:
非常感谢回复！我遇到的问题是需要一个同时带有GPU驱动和RT内核的系统，orangepi原厂的镜像有一个RT内核可以直接用，但是没有支持panfrost驱动，第三方的镜像有panfrost但是又没有RT内核，而且内核都是经过修改适配的，linux官方的RT补丁也不能用，所以想先尝试一下给原厂的镜像安装panfrost驱动，但是按那些方法都没能成功，我怀疑是x11的问题，请教一下有什么好的建议么？你有在x11的系统上直接安装成功过吗？
 

251221597
***@***.***

eternal-Eason, 2024-09-24:
@wanyingwe  已经不用内核里面panfrost驱动了。只要闭源gpu驱动可以用，开源的Panfork 也可以用。也就是说，你直接拿rk内核来用，不经过任何修改内核配置，Panfork可以正常起来，rk3588目前是这样。orangepi原厂的镜像 的gpu可以用吗？也就是glmark2 测试显示gpu渲染还是cpu渲染的

wanyingwe, 2024-09-24:
@eternal-Eason 我用的也是3588的板子，最近rockchip的内核有更新吗？orangepi原厂的镜像只支持了opengl-es，不支持标准版opengl，跑glmark2的话显示的是软件渲染

eternal-Eason, 2024-09-24:
@wanyingwe glmark2-es2-drm或者其他glmark2-es2-xxx呢？闭源gpu测试glmark2会显示cpu渲染。

wanyingwe, 2024-09-24:
@eternal-Eason  glmark2-es是可以用的，可以用GPU硬件加速，但是有一些应用识别不到opengl-es的驱动，比如chromium和firefox，都无法使用硬件加速，只有rockchip提供的修改过的chromium可以使用GPU加速，但不知道他们改了些什么东西。panfork的支持就好多了，大部分的应用就都能正常使用GPU加速了