Joshua-Riek/ubuntu-rockchip #790: NanoPC T6 HDMI0 Audio Left Channel stop working

Spirarementum, 2024-05-11:
First allow me to thank you for the great work, time, and devotion.

**Issue:**
The left HDMI0 Channel stopped working today.
Both audio channel works on the HSMI1.
The only thing that changed was a system update and a manual reboot.

I am fishing here trying to diagnose the problem...Could this be linked to the last system update ?

Let me know if I can provide more info to help pinpoint the problem.

wmport, 2024-05-11:
Everything works, check your hdmi cable

Joshua-Riek, 2024-05-11:
What OS are you running?

Spirarementum, 2024-05-11:
Just tested the cable.
Also tested on another monitor to rule out the hardware.
The HDMI0 left audio output from the NanoPC T6 stopped working right after the last update (yesterday).
Running Ubuntu 22.04LTS

wmport, 2024-05-12:
I launched 22.04 on an SD card, with the latest update, on hdmi0 and hdmi1 all audio channels work.

closed 2024-05-12 by Joshua-Riek

Spirarementum, 2024-05-12:
Thank you.
Again I have to compliment you on your work....Ubuntu 22.04 is working flawlessly on the nanopc-t6 :)