Joshua-Riek/ubuntu-rockchip #793: Cannot compile for board orange pi 5 plus

pugant, 2024-05-11:
Hello,
i'm new to ARM world and i'd want to compile myself a img file for my orangepi-5-plus:

I'd want to set CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y in kernel parameters because i'd want to install clamav with clamonacc feature on my orange pi 5 plus.
Was my first time that i try to compile myself the linux kernel and for setting up environment for build, i set up a vm with Ubuntu 22.04 following instruction here: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-(old)
After launch (sudo ./build.sh  --board=orangepi-5-plus --release=noble --project=desktop) script remain stuck with this error message:
2024/05/11 11:34:25.195300 helpers.go:150: error trying to compare the snap system key: system-key missing on disk

Am I doing something wrong? (probably!! :-))
[build-20240511104723.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15283715/build-20240511104723.log)
Thanks Joshua for your amazing work! <3

Joshua-Riek, 2024-05-11:
I noticed this error too, i'm trying to fix it, previously i have ignored the error.

But if you only need to modify a kernel config option there is no reason to build an image.

pugant, 2024-05-11:
Ok, thanks for suggestion, i'll try to compile only the linux kernel as a deb package and update it directly on board. 
I'll follow instruction here: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-(old)

A curiosity:
There is a specific reason (e.g. performance issues, ...) why for rockchip only, the CONFIG_FANOTIFY_ACCESS_PERMISSIONS variable is not set to y

root@ubuntu-orangepi:/home/pugant/Downloads/linux-rockchip# grep -r CONFIG_FANOTIFY_ACCESS_PERMISSIONS
arch/arm64/configs/defconfig:CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y
arch/arm64/configs/rockchip_defconfig:# CONFIG_FANOTIFY_ACCESS_PERMISSIONS is not set
arch/loongarch/configs/loongson3_defconfig:CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y
arch/s390/configs/defconfig:CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y

Joshua-Riek, 2024-05-11:
> Ok, thanks for suggestion, i'll try to compile only the linux kernel as a deb package and update it directly on board. 
I'll follow instruction here: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-(old)

Those kernel compile instructions are likely out of date especially if you are using ubuntu 24.04. 

> A curiosity:
There is a specific reason (e.g. performance issues, ...) why for rockchip only, the CONFIG_FANOTIFY_ACCESS_PERMISSIONS variable is not set to y
> root@ubuntu-orangepi:/home/pugant/Downloads/linux-rockchip# grep -r CONFIG_FANOTIFY_ACCESS_PERMISSIONS
arch/arm64/configs/defconfig:CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y
arch/arm64/configs/rockchip_defconfig:# CONFIG_FANOTIFY_ACCESS_PERMISSIONS is not set
arch/loongarch/configs/loongson3_defconfig:CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y
arch/s390/configs/defconfig:CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y

The kernel comes from Android sources, so I could not enable typical Linux drivers as a large portion are broken. However if it works you can send a pr to the kernel or ask for it to be enabled.

pugant, 2024-05-12:
Just for alignment... 
Using a vm on kvm, with Ubuntu 22.04 LTS.

1. git clone -b noble https://github.com/Joshua-Riek/linux-rockchip.git
2. (modified **arch/arm64/configs/rockchip_defconfig** and **debian.rockchip/config/config.common.ubuntu** with setting **CONFIG_FANOTIFY_ACCESS_PERMISSIONS=y**)
3. make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 rockchip_linux_defconfig
4. make KBUILD_IMAGE="arch/arm64/boot/Image" CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 -j "$(nproc)" bindeb-pkg
5. These are the files produced by the build process:
      - linux-headers-6.1.43-gc406baccc202-dirty_6.1.43-gc406baccc202-dirty-1_arm64.deb
      - linux-image-6.1.43-gc406baccc202-dirty_6.1.43-gc406baccc202-dirty-1_arm64.deb
      - linux-image-6.1.43-gc406baccc202-dirty-dbg_6.1.43-gc406baccc202-dirty-1_arm64.deb
      - linux-libc-dev_6.1.43-gc406baccc202-dirty-1_arm64.deb
6. Transferred *.deb files on orangepi5plus with **ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-plus.img** booted from  nvme storage
7. installed using dpkg -i *.deb

The board don't start after update (Led remain red)

Joshua-Riek, 2024-06-28:
This has been enabled, in the future it would be helpful for someone to send a pull request, its hard to keep track of everything. 

ref: https://github.com/Joshua-Riek/linux-rockchip/commit/9d668cb81ee72be4ac336cc734c56dee630ba32f
ref: https://github.com/Joshua-Riek/linux-rockchip/commit/be8fa3c2d98532a59d4342d310b3a04d80d4cd4a

closed 2024-06-28 by Joshua-Riek