Joshua-Riek/ubuntu-rockchip #795: Ubuntu Server 22.04 - on Orange Pi 3B - Load average greater than 1 on idle system with fresh install

abasu0713, 2024-05-11:
# Summary
Just trying to understand why this is happening. I have constant load average of greater than 1.0 on a fresh Ubuntu Server (Jammy) install Orange Pi 3B. Using this [release](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.33)

```bash
ubuntu@ubuntu:~$ uptime
 20:17:11 up 21 min,  2 users,  load average: 1.16, 1.03, 0.87
ubuntu@ubuntu:~$ top

top - 20:17:46 up 21 min,  2 users,  load average: 1.23, 1.06, 0.89
Tasks: 150 total,   1 running, 149 sleeping,   0 stopped,   0 zombie
%Cpu(s):  3.2 us,  6.5 sy,  0.0 ni, 90.3 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7686.6 total,   7205.7 free,    190.0 used,    290.9 buff/cache
MiB Swap:   1024.0 total,   1024.0 free,      0.0 used.   7400.3 avail Mem 
```

Sample snapshot of `top`:
[image]

would appreciate any help. 
> This happens on the latest current kernel 6.6 with OrangePi's official Ubuntu Server image as well.

thewonderer73, 2024-05-16:
I read somewhere here that it's due to 22.04.  24.04 resolves this load average issue.   Give 24.04 a try.

abasu0713, 2024-05-18:
> I read somewhere here that it's due to 22.04. 24.04 resolves this load average issue. Give 24.04 a try.

@thewonderer73 No good. Still the same

```bash
ubuntu@ubuntu:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
ubuntu@ubuntu:~$ uptime
 22:44:04 up 10 min,  1 user,  load average: 1.09, 1.20, 0.74
ubuntu@ubuntu:~$ top

top - 22:44:47 up 10 min,  1 user,  load average: 1.04, 1.17, 0.75
Tasks: 466 total,   1 running, 465 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.2 us,  0.2 sy,  0.0 ni, 99.3 id,  0.2 wa,  0.0 hi,  0.1 si,  0.0 st 
MiB Mem :   7685.7 total,   6662.1 free,    360.0 used,    759.4 buff/cache     
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   7325.7 avail Mem 
```

I am seeing these hundreds of worker threads:
[image]

abasu0713, 2024-05-19:
I can also confirm that this issue is non-existent on the latest kernel version for Armbian with Debian - Bookworm flavor. 

[image]

Closing this, because who needs Ubuntu! lol

closed 2024-05-19 by abasu0713