Joshua-Riek/ubuntu-rockchip #796: GPU rendering artifacts, corruption, performance degradation after recent 22.04 update

aubreyrjones, 2024-05-11:
I took an update including new kernels and firmware today.

After the update, I am noticing serious artifacts in 3d applications. For instance, viewing the three.js examples in chromium at `https://threejs.org/examples/#webgl_animation_keyframes` is seriously broken. Especially after several reloads of the page or other 3d applications.

I am using X. (Wayland is not compatible with my accessibility applications or window manager.)

In the short run, is it safe to downgrade kernel and firmware?

How could I contribute to resolving the issue? I have a lot of low-level/embedded experience, but I don't know where to start with something so immense as a linux distro.

Joshua-Riek, 2024-05-11:
The GPU driver Panfork is dead and will not be worked on ever again. There really is no solution unless Rockchip decides to fully support the upstream Panthor GPU driver in their BSP kernel. Even then, the Panthor driver preforms worse and has a lot of bugs that have not been addressed. Additionally, the Panfork developer is hostile to multiple Mesa developers, some of his hostilities hit close to home for me. Including his work hurts, and ii will not think about debugging or patching it myself for a second, but it's really the closest option we have to something useable.

Since I've been working on these boards for over a year now, I'm trying to get direct contact with Rockchip and see what I can do to advocate for better GPU support, but it takes time and I'm honestly a random person on the internet.

aubreyrjones, 2024-05-11:
@Joshua-Riek I guess I can understand your position, but could you help me understand the technology?

I see that there is a MESA driver Panfrost with ARM's blessing. How does this relate to Panfork? I see people calling Panfork a "userspace hack", but I don't understand what it is overall. Is it this ROCKNIX mesa fork I found? I see that you're pulling in a build of it from a PPA you control, but I don't see a github repo for it on your page? Are you patching that, or is it just as the hostile developer shipped?

Can I just fork the fork and then be less hostile or whatever?

I can empathize with your distaste, even if I don't understand it. But can you help me to take over that burden for you? I am very much interested in getting half-decent GPU support on the Orange Pi 5. In particular, I am developing an AR+sensors application on it--and also using the OPi5 as my main workstation these days. I am happy to roll up my sleeves and help with some things you find distasteful. If you can give me an idea how to build panfork in such a way that I can actually load it via apt, that would be enough to get me started.

Also, I am only reporting this issue because things changed between updates. I did not have the artifacts before the update. I took the update, and now have the artifacts. Could you please start by giving me an idea of what changed in the kernel and firmware? Is it safe to downgrade kernel and firmware on these boards? Do I need to downgrade uboot as well?

aubreyrjones, 2024-05-11:
After more research... is the issue less actual kernel support, and more that Rockchip is lagging on board support?

Joshua-Riek, 2024-05-12:
I'm sorry for not providing detailed information, but I'm extremely tired and caring less and less. I have been trying to get the RK3588 platform in a more usable state for a while now, as many board vendors and Rockchip are lagging behind on software support. People do not realize I have no corporate sponsorships or deals and that I'm a random person on the internet, yet I develop and support more than 20 boards with two Ubuntu LTS releases. I've invested way too much of my personal time into the Rockchip platform and I'm now feeling the effects of it. 

I'm simply not in a state where I could be helpful, my current plan is to get the next release out tonight and then take a step back for my health and mental wellbeing. I would recommend giving this upcoming release a try as I have completely re-worked Ubuntu 22.04 and the 5.10 kernel, it may solve your problem.

aubreyrjones, 2024-05-12:
I completely understand. I'm struggling with burnout myself. And for what it's worth, I know you're doing this on your own. Lots of people linking to your work, but I don't see a lot of contributors.

I wish you could mentor me in maintaining this distro, but I'm hearing that you don't have the bandwidth for that. No problem. I hope that stepping back gives you the space you need.

develperbayman, 2024-05-13:
sounds like y'all need to take a break ....please take a break ....a good long one .....if you quit we are boned so you must avoid burnout

MarkhamLee, 2024-05-13:
@Joshua-Riek I'd just like to second when @bagong said: I greatly appreciate all the work you do on these boards, my and a lot of people's 3588 devices would be gathering dust without your Ubuntu distro. 

The manufacturers really need to step up to support you, I don't think they understand that the viability of this platform requires supporting if not hiring teams of devs like yourself. Rockchip and Cannonical really need to look at this community and have a conversation. In this age of AI/ML engineers need more options than hard to find and expensive NVIDIA devices and under powered Raspberry Pis 

Finally, I'd like to help if I can. I'm a data & ML engineer so my knowledge of embedded programming is limited, is there anything you can point myself (and others too) so we can skill up and assist?

Joshua-Riek, 2024-05-14:
I can't express how much some of these messages mean to me. All I can say is thank you. I will be logging out for at least 2 weeks; I will come back, but I'm taking some time for my health. 

Thanks again,

Joshua RIek

StuartIanNaylor, 2024-05-15:
@aubreyrjones 
> I see that there is a MESA driver Panfrost with ARM's blessing. How does this relate to Panfork? I see people calling Panfork a "userspace hack", but I don't understand what it is overall.

Panfork began when the G610 was not supported and is the pre Panthor stuff to support Valhall specifically for the G610 with the Rockchip BSP.
Panfrost and the current one set to work with the latest RC 6.10 should be avail mainline, but that is between Arm and Collabora
So its no-mans land with newer BSP kernels as panfork has stopped and the 1st Vallhall implemented Panfrost will be the 6.10 release that is still waiting for a RC1 but it is the next release.

@Joshua-Riek After your break do you think you might be releasing mainline-experimental of the 6.10 rc's?

closed 2024-12-16 by github-actions[bot]