Joshua-Riek/ubuntu-rockchip #797: i386 Package Install Error

megasouyutubers, 2024-05-12:
Hello, I am trying to install i386 package but there is an error. I need this package to install SteamCMD.

`steam@ubuntu~steamcmd$ sudo dpkg --add-architecture i386
sudo apt update
sudo apt install libc6i386 libstdc++6i386
Hit1 httpports.ubuntu.comubuntu-ports noble InRelease
Hit2 httpppa.launchpad.netjjriekrockchipubuntu noble InRelease
Get3 httpports.ubuntu.comubuntu-ports noble-updates InRelease [89.7 kB]
Hit4 httpppa.launchpad.netjjriekrockchip-multimediaubuntu noble InRelease
Hit5 httpppa.launchpad.netjjriekpanfork-mesaubuntu noble InRelease
Hit6 httpports.ubuntu.comubuntu-ports noble-backports InRelease
Hit7 httpports.ubuntu.comubuntu-ports noble-security InRelease
Ign8 httpports.ubuntu.comubuntu-ports noblemain i386 Packages
Ign9 httpports.ubuntu.comubuntu-ports nobleuniverse i386 Packages
Ign10 httpports.ubuntu.comubuntu-ports noblerestricted i386 Packages
Ign11 httpports.ubuntu.comubuntu-ports noblemultiverse i386 Packages
Ign8 httpports.ubuntu.comubuntu-ports noblemain i386 Packages
Ign9 httpports.ubuntu.comubuntu-ports nobleuniverse i386 Packages
Ign10 httpports.ubuntu.comubuntu-ports noblerestricted i386 Packages
Ign11 httpports.ubuntu.comubuntu-ports noblemultiverse i386 Packages
Ign8 httpports.ubuntu.comubuntu-ports noblemain i386 Packages
Ign9 httpports.ubuntu.comubuntu-ports nobleuniverse i386 Packages
Ign10 httpports.ubuntu.comubuntu-ports noblerestricted i386 Packages
Get12 httpppa.launchpad.netjjriekrockchipubuntu noblemain i386 Packages [1816 B]
Ign11 httpports.ubuntu.comubuntu-ports noblemultiverse i386 Packages
Ign8 httpports.ubuntu.comubuntu-ports noblemain i386 Packages
Ign9 httpports.ubuntu.comubuntu-ports nobleuniverse i386 Packages
Ign10 httpports.ubuntu.comubuntu-ports noblerestricted i386 Packages
Ign11 httpports.ubuntu.comubuntu-ports noblemultiverse i386 Packages
Ign8 httpports.ubuntu.comubuntu-ports noblemain i386 Packages
Ign9 httpports.ubuntu.comubuntu-ports nobleuniverse i386 Packages
Ign10 httpports.ubuntu.comubuntu-ports noblerestricted i386 Packages
Ign11 httpports.ubuntu.comubuntu-ports noblemultiverse i386 Packages
Err8 httpports.ubuntu.comubuntu-ports noblemain i386 Packages
  404  Not Found [IP 26202d4000119 80]
Get13 httpppa.launchpad.netjjriekrockchip-multimediaubuntu noblemain i386 Packages [3480 B]
Ign9 httpports.ubuntu.comubuntu-ports nobleuniverse i386 Packages
Ign10 httpports.ubuntu.comubuntu-ports noblerestricted i386 Packages
Ign11 httpports.ubuntu.comubuntu-ports noblemultiverse i386 Packages
Get14 httpppa.launchpad.netjjriekpanfork-mesaubuntu noblemain i386 Packages [688 B]
Ign15 httpports.ubuntu.comubuntu-ports noble-updatesmain i386 Packages
Ign16 httpports.ubuntu.comubuntu-ports noble-updatesuniverse i386 Packages
Ign15 httpports.ubuntu.comubuntu-ports noble-updatesmain i386 Packages
Ign16 httpports.ubuntu.comubuntu-ports noble-updatesuniverse i386 Packages
Ign15 httpports.ubuntu.comubuntu-ports noble-updatesmain i386 Packages
Ign17 httpports.ubuntu.comubuntu-ports noble-backportsuniverse i386 Packages
Ign16 httpports.ubuntu.comubuntu-ports noble-updatesuniverse i386 Packages
Ign15 httpports.ubuntu.comubuntu-ports noble-updatesmain i386 Packages
Ign17 httpports.ubuntu.comubuntu-ports noble-backportsuniverse i386 Packages
Ign16 httpports.ubuntu.comubuntu-ports noble-updatesuniverse i386 Packages
Ign15 httpports.ubuntu.comubuntu-ports noble-updatesmain i386 Packages
Ign17 httpports.ubuntu.comubuntu-ports noble-backportsuniverse i386 Packages
Ign16 httpports.ubuntu.comubuntu-ports noble-updatesuniverse i386 Packages
Err15 httpports.ubuntu.comubuntu-ports noble-updatesmain i386 Packages
  404  Not Found [IP 26202d4000119 80]
Ign17 httpports.ubuntu.comubuntu-ports  noble-backportsuniverse i386 Packages
Ign18 httpports.ubuntu.comubuntu-ports noble-securitymain i386 Packages
Ign19 httpports.ubuntu.comubuntu-ports noble-securityuniverse i386 Packages
Ign16 httpports.ubuntu.comubuntu-ports noble-updatesuniverse i386 Packages
Ign17 httpports.ubuntu.comubuntu-ports noble-backportsuniverse i386 Packages
Ign18 httpports.ubuntu.comubuntu-ports noble-securitymain i386 Packages
Ign19 httpports.ubuntu.comubuntu-ports noble-securityuniverse i386 Packages
Err17 httpports.ubuntu.comubuntu-ports noble-backportsuniverse i386 Packages
  404  Not Found [IP 26202d4000119 80]
Ign18 httpports.ubuntu.comubuntu-ports noble-securitymain i386 Packages
Ign19 httpports.ubuntu.comubuntu-ports noble-securityuniverse i386 Packages
Ign18 httpports.ubuntu.comubuntu-ports noble-securitymain i386 Packages
Ign19 httpports.ubuntu.comubuntu-ports noble-securityuniverse i386 Packages
Ign18 httpports.ubuntu.comubuntu-ports noble-securitymain i386 Packages
Ign19 httpports.ubuntu.comubuntu-ports noble-securityuniverse i386 Packages
Err18 httpports.ubuntu.comubuntu-ports noble-securitymain i386 Packages
  404  Not Found [IP 26202d4000119 80]
Ign19 httpports.ubuntu.comubuntu-ports noble-securityuniverse i386 Packages
Fetched 95.7 kB in 2s (53.4 kBs)
Reading package lists... Done
E Failed to fetch httpports.ubuntu.comubuntu-portsdistsnoblemainbinary-i386Packages  404  Not Found [IP 26202d4000119 80]
E Failed to fetch httpports.ubuntu.comubuntu-portsdistsnoble-updatesmainbinary-i386Packages  404  Not Found [IP 26202d4000119 80]
E Failed to fetch httpports.ubuntu.comubuntu-portsdistsnoble-backportsuniversebinary-i386Packages  404  Not Found [IP 26202d4000119 80]
E Failed to fetch httpports.ubuntu.comubuntu-portsdistsnoble-securitymainbinary-i386Packages  404  Not Found [IP 26202d4000119 80]
E Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E Unable to locate package libc6i386
E Unable to locate package libstdc++6i386`

Joshua-Riek, 2024-05-12:
Please provide the exact steps so this is reproducible. Also, have you setup multiarch?

megasouyutubers, 2024-05-12:
I have setup multiarch and i get the error 404 (from previous text) when i execute "sudo apt update && sudo apt install build-essential git cmake gcc-arm-linux-gnueabihf libc6:armhf libc6-dev-armhf-cross libappindicator1 libncurses5:armhf libstdc++6:armhf libnm0 libtcmalloc-minimal4 binfmt-support liblttng-ust-dev libcairo2:armhf libgmp10:armhf libvulkan1:armhf libudev-dev:armhf -y"

Also i got the orange pi 5 around 1 week ago and i dont know a lot about ubuntu so let me know if you need any more information.
```
sudo apt update & sudo apt upgrade -y

sudo apt install curl

sudo dpkg --add-architecture armhf

sudo apt update && sudo apt install build-essential git cmake gcc-arm-linux-gnueabihf libc6:armhf libc6-dev-armhf-cross libappindicator1 libncurses5:armhf libstdc++6:armhf libnm0 libtcmalloc-minimal4 binfmt-support liblttng-ust-dev libcairo2:armhf libgmp10:armhf libvulkan1:armhf libudev-dev:armhf -y

git clone 

cd box86 

mkdir build; cd build; cmake .. -DRK3588=1 -DCMAKE_BUILD_TYPE=RelWithDebInfo

make -j4

sudo make install

cd ~

git clone 

cd box64

mkdir build; cd build; cmake .. -DRK3588=1 -DCMAKE_BUILD_TYPE=RelWithDebInfo
make -j4

sudo make install

sudo useradd steam-m

sudo -u steam-s

cd ~
mkdir steamcmd

cd steamcmd

curl -sqL "https://steamcdn-a.akamaihd.net/client/installer/steamcmd_linux.tar.gz" | tar zxvf -
./steamcmd.sh
```

closed 2024-06-26 by megasouyutubers