Joshua-Riek/ubuntu-rockchip #798: NVMe installaion error with v2.1.0

adr-bed, 2024-05-13:
Hello!
I get error for script `ubuntu-rockchip-install` when trying to install the system to the nvme drive. The error appears on a fresh installation of Ubuntu Server v2.1.0 for Rock 5B:

```
Creating partition table for /dev/nvme0n1.
Creating filesystem on partitions /dev/nvme0n1p1 and /dev/nvme0n1p2.

findmnt: unknown column: SOURCES
Error: in /usr/bin/ubuntu-rockchip-install on line 187
```

Joshua-Riek, 2024-05-13:
Which OS version? Ubuntu 24.04 or Ubuntu 22.04?

adr-bed, 2024-05-13:
> Which OS version? Ubuntu 24.04 or Ubuntu 22.04?

24.04

ErcinDedeoglu, 2024-05-20:
hit same problem.

> apt update
> apt upgrade

now working as expected. thanks!

lucvanbraekel, 2024-05-24:
I encountered the same problem today when executing ubuntu-rockchip-install on Ubuntu server 22.04 for Orange Pi 5.
Yesterday I executed the same for 24.04 and everything went fine there.

Joshua-Riek, 2024-06-06:
This has been fixed, update the system with apt.

closed 2024-06-06 by Joshua-Riek