Joshua-Riek/ubuntu-rockchip #799: armsom-sige7 cannot boot from eMMC

rcarmo, 2024-05-13:
I installed from an SD card and copied the installation to eMMC, but the board refuses to boot from eMMC unless I hold `RESET` upon power up.

There's no 'mtd' device, so installing uBoot to the SPI fails:

```bash
sudo u-boot-install-mtd
[sudo] password for me: 
/usr/bin/u-boot-install-mtd: block device '/dev/mtd0' not found
/usr/bin/u-boot-install-mtd: writing U-Boot to the SPI flash
While trying to open /dev/mtd0 for read/write access: No such file or directory
Error: in /usr/bin/u-boot-install-mtd on line 21
```

Any suggestions? I'm used to `armbian-config` having a way to toggle boot order in the SPI in similar boards, but cannot find any references to similar functionality in this image.

aubreyrjones, 2024-05-13:
I don't know about that particular board, but on the Orange Pi 5, the SPI memory is not called `mtd0`. It's like `mmtdblk0` or something like that . If you have a running system, you can find it by doing `lsblk`. You can then pass that as an argument to `u-boot-install-mtd` for the install.

Not sure if that will solve your issue, but I think it'll get you closer.

rcarmo, 2024-05-13:
Well, an `lsblk` shows me 2 4MB devices that are... suspicious.

This is after managing to boot from the eMMC, nothing else connected:

```bash
~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0  93.7M  1 loop /snap/lxd/28474
loop1          7:1    0  69.2M  1 loop /snap/core22/1383
loop2          7:2    0  33.7M  1 loop /snap/snapd/21467
loop3          7:3    0  92.8M  1 loop /snap/lxd/28475
mmcblk0      179:0    0 115.3G  0 disk 
└─mmcblk0p1  179:1    0 115.2G  0 part /
mmcblk0boot0 179:32   0     4M  1 disk 
mmcblk0boot1 179:64   0     4M  1 disk 
```

They don't look like MTD devices though (I've dealt with 3 or 4 different RockChip boards). They don't show up on `lshw`, for starters.

They do have different device nodes:

```bash
$ ls -al /dev/mmcblk0*
brw-rw---- 1 root disk 179,  0 May 13 16:09 /dev/mmcblk0
brw-rw---- 1 root disk 179, 32 May 13 16:08 /dev/mmcblk0boot0
brw-rw---- 1 root disk 179, 64 May 13 16:08 /dev/mmcblk0boot1
brw-rw---- 1 root disk 179,  1 May 13 16:09 /dev/mmcblk0p1
crw------- 1 root root 235,  0 May 13 16:08 /dev/mmcblk0rpmb
```

Still, rather than trying to blindly flash stuff (I've had enough tussles with Rockchip recovery), I'm wondering if anyone has a definitive answer.

Joshua-Riek, 2024-05-13:
Hello, what Ubuntu version and how are you installing the OS onto the eMMC?

rcarmo, 2024-05-13:
24.04, and I used the instructions on the Wiki.

Joshua-Riek, 2024-05-13:
It works fine for me, using the command `sudo ubuntu-rockchip-install /dev/mmcblk0`.

rcarmo, 2024-05-13:
I did that as well, but the device won't boot from eMMC unless I hold down the RESET button and hit power.

Joshua-Riek, 2024-05-13:
What type of power supply are you using? I know this board has some problems with PD charging so i use a dumb 5v 4a power supply.

rcarmo, 2024-05-13:
I am using an Anker charger - however, this did not happen until I flashed your image (I had Armbian running fine).

I managed to hook up an FTDI adapter to the serial console, and I am getting a really weird output:

Straight upon power on:

```bash
DDR V1.11 f1474cf52f cym 23/05/09-11:02:36
unknown device
If no 'may be ch* soldering abnormality' is printed, it may be ch0 soldering abnormality
error
ERR
```

...and then it stays there. When I hit RESET, I get the same message, but _sometimes_ it goes and boots:

```bash
DDR V1.11 f1474cf52f cym 23/05/09-11:02:36
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=16 Size=4096MB
Manufacturer ID:0xff
CH0 RX Vref:28.9%, TX Vref:21.8%,20.8%
CH1 RX Vref:28.9%, TX Vref:19.8%,19.8%
CH2 RX Vref:30.5%, TX Vref:18.8%,19.8%
CH3 RX Vref:31.4%, TX Vref:18.8%,18.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09 (Apr 27 2024 - 16:57:13)
unknown raw ID 0 0 0
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
MMC: no card present
mmc_init: -123, time 1
spl: mmc init failed with error: -123
Trying to boot from MMC1
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(7efcd01a0f...) + OK
## Checking uboot 0x00200000 ... sha256(4f20ff71b2...) + OK
## Checking fdt 0x00323b00 ... sha256(3ad41d25ce...) + OK
## Checking atf-2 0xff100000 ... sha256(1163474a5b...) + OK
## Checking atf-3 0x000f0000 ... sha256(da90adf3a4...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 115.786 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-589-g3389cfdda:derrick.huang
NOTICE:  BL31: Built : 10:14:29, May  9 2023
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
[... 132 lines cut ...]
39186944 bytes read in 237 ms (157.7 MiB/s)
append: root=UUID=09f2e57a-8a8c-4486-94fe-e91519305a22 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles
Retrieving file: /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/rk3588-armsom-sige7.dtb
271139 bytes read in 28 ms (9.2 MiB/s)
Fdt Ramdisk skip relocation
No misc partition
## Flattened Device Tree blob at 0x08300000
   Booting using the fdt blob at 0x08300000
  'reserved-memory' ramoops@110000: addr=110000 size=e0000
   Using Device Tree in place at 0000000008300000, end 0000000008345322
Adding bank: 0x00200000 - 0xf0000000 (size: 0xefe00000)
Adding bank: 0x100000000 - 0x3fc000000 (size: 0x2fc000000)
Adding bank: 0x3fc500000 - 0x3fff00000 (size: 0x03a00000)
Adding bank: 0x4f0000000 - 0x500000000 (size: 0x10000000)
Total: 5900.357 ms

Starting kernel ...

[    6.231154] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    6.231182] Linux version 6.1.0-1013-rockchip (buildd@bos01-arm64-013) (aarch64-linux-gnu-gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #13-Ubuntu SMP Thu May  9 21:20:56 UTC 2024
[    6.238585] Machine model: ArmSoM Sige7
...
```

So either I have a dodgy RAM chip all of a sudden, or there's something I'm missing here.

Joshua-Riek, 2024-05-13:
I changed the original U-Boot code that was submitted as a pull request as it affected other devices while a better solution was available with a little extra work (there could be a problem with my bootloader changes, but I have some confidence in my bootloader code).

 But the below output interests me, it's not something I have seen before. Let me do some digging and see if i can find anything.
 ```
 DDR V1.11 f1474cf52f cym 23/05/09-11:02:36
unknown device
If no 'may be ch* soldering abnormality' is printed, it may be ch0 soldering abnormality
error
ERR
 ```

Joshua-Riek, 2024-05-13:
Hmm after a quick grep in the bootloader source code the above message seems to come from the DDR or SPL blob which is sadly closed source, this tracks with your suspicion of doggy RAM chips. 

However, I'm using older DDR and SPL blobs I will update them shortly and start a build with GitHub actions, hopefully it will solve this issue.

Joshua-Riek, 2024-05-13:
Hmm Armbian uses the same DDR and SPL blobs as me, so this is very strange.

rcarmo, 2024-05-14:
Maybe there is some kind of default setting that needs to be reset?

rcarmo, 2024-07-07:
I re-flashed uboot and this went away magically.

closed 2024-07-07 by rcarmo

dawndrums, 2025-05-04:
> I re-flashed uboot and this went away magically.

which uboot did you reflash? I am facing the same issue with one of my custom boards. When I replaced the SoC, it went away. Then again it started showing up again, and now it's permanent. I feel the first SoC was fine and also the second replacement. I also don't suspect the RAM somehow. Could be a power supply issue. Can you reproduce it, when flashing back ubuntu?

mathansen, 2026-01-06:
I just ran into the same issue on an OrangePi 5 Ultra. After entering maskrom mode, I realised that the "soldering abnormality" error was also thrown when trying to flash the bootloader. As a Hail Mary solution, I went with the resolder approach suggested by @royka1, drenched the LPDDR in flux and pointed my heat gun at it for a few seconds. Afterward, I was able to reflash https://dl.radxa.com/rock5/sw/images/loader/rock-5b/release/rk3588_spl_loader_v1.15.113.bin without any issues.