Joshua-Riek/ubuntu-rockchip #800: Orange Pi 5 Plus w/ 24.02 and LCD Overlay

anernest, 2024-05-13:
- Orange Pi 5 Plus w/32GB & 256GB eMMC
- Orange Pi 5 Plus 10.1" Touch LCD
- Installed 24.02 to eMMC per instructions
- Add the orangepi-5-plus-lcd.dtbo overlay file to /etc/default/u-boot per instructions
- sudo u-boot-update

> P: Checking for EXTLINUX directory... found.
> P: Writing config for vmlinuz-6.1.0-1013-rockchip...
> P: Writing config for vmlinuz-6.1.0-1012-rockchip...
> P: Updating /boot/extlinux/extlinux.conf...

- sudo reboot
- red LED flashes for a bit
- then alternates red/white?
- then solid red
- nic LEDs flash
- no activity on either HDMI attached nor 
- Nothing more (have left for over 15min)

Joshua-Riek, 2024-05-13:
Going to need UART logs, the LCD display is not something I've tested or have. But from what I know display panels are very broken with the 6.1 kernel. So, I'd recommend trying Ubuntu 22.04.

anernest, 2024-05-13:
Not sure how to go about getting the UART logs - the only way I know how to recover is to re-install the OS.

I'll switch back to 22.04 - I think I saw someone said they got the LCD working with it.

I'll close as not planned.

BTW, very impressive body of work.

anernest, 2024-05-13:
...

anernest, 2024-05-13:
...

anernest, 2024-05-13:
...

closed 2024-05-13 by anernest