Joshua-Riek/ubuntu-rockchip #802: Missing partition table

ohyeah521, 2024-05-16:
build command:
`sudo ./build.sh --board=rock-5b --release=jammy --project=server --kernel=rockchip-5.10`

There were no errors during the process, and the image file was generated, but when opening the image with balenaEtcher, it prompted that there was no partition table.

closed 2024-05-22 by Joshua-Riek

ohyeah521, 2024-06-06:
@Joshua-Riek this is build log
[build-20240606171049.log](https://github.com/user-attachments/files/15629928/build-20240606171049.log)
[build-20240606174936.log](https://github.com/user-attachments/files/15629938/build-20240606174936.log)

[image]

ohyeah521, 2024-06-07:
@Joshua-Riek  Maybe this is the reason. The file name generated after compilation is ubuntu-22.04-preinstalled-server-arm64.rootfs.tar. The file in the compressed package is not img, but the content in the picture below.
[image]

Joshua-Riek, 2024-06-07:
That is the expected behavior.