Joshua-Riek/ubuntu-rockchip #806: Flash Image to eMMC of NanoPC-T6 (non-LTS version)

rdu-weston, 2024-05-17:
Hi, I've been using the Ubuntu 22.04 from this project for some time and I just tried to upgrade to the latest v1.33 release. I can successfully build the image and boot from a microsd card (after flashing the image using etcher and then install uboot to it). However, it seems I cannot make the system bootable from eMMC.

What I have tried:

1. from a previous release of this project, flash the image from a microSD card using:

```
$ xz -dc ubuntu-22.04.3-preinstalled-desktop-arm64-nanopc-t6.img.xz | sudo dd of=/dev/mmcblk0 bs=4k
$ sync
```

2. flash the bootloader to eMMC and then flash the image using:

```
$ sudo u-boot-install /dev/mmcblk0
$ sudo ubuntu-rockchip-install /dev/mmcblk0
```

Neither method worked for me. BTW, my NanoPC-T6 is the non-LTS version with the SPI flash chip. I also noticed major changes in the wiki of the project. I was wondering what has been changed in the boot process and why the installation and configurations look so different now. Any suggestions?

Thank you!

rdu-weston, 2024-05-17:
I thought v2 release was only for Ubuntu 24.04 According to release notes of the latest release,I was wrong apparently. I will try to regenerate images from the current release and try the latest wiki instructions again. I think it would be great if the original wiki pages are kept for older images (maybe marked as old as well). Now they are mostly gone.

wmport, 2024-05-18:
Check the spelling of the image you are trying to install from the command line.

rdu-weston, 2024-05-20:
> Check the spelling of the image you are trying to install from the command line.

I don't see any spelling errors, can you be more specific? Thanks.

wmport, 2024-05-20:
You specified above "ubuntu-22.04.3-preinst..."
Latest Issue Images
ubuntu-22.04-....
ubuntu-24.04-....

Run from a directory with a short name

rdu-weston, 2024-05-20:
I see what you meant. I built the image locally with modifications to the kernel and the name I specified was the name of the image I got from the build process.

rdu-weston, 2024-05-20:
I just tested the image generated with release v2.1.0 and it works by following instructions on the new wiki for 22.04. Closing the issue now.